CTCaer/hekate #344: reboot_payload.bin keep getting overwrrtten

ELY3M, 2020-01-16:
I am trying to set up a dual boot system on my friend's switch.  
I and my friend want to keep atmosphere reboot_payload.bin original one from atmosphere GitHub.    
if I boot to OFW then I come back to CFW and I use reboot to payload in hbmenu.  
I was expecting that atmosphere to be rebooted to atmosphere.   
I saw hekate menu and I checked the sd card....  I kept replacing the reboot_payload.bin  

I looked in the code...  I found the function in  /bootloader/fss.c    
why do it copy to reboot_payload.bin?        

look at this hekate ini that I am using.    

[config]
autoboot=0
autoboot_list=0
bootwait=3
customlogo=1
verification=2
backlight=160
autohosoff=0
autonogc=0
[Atmosphere Test]
payload=atmosphere/reboot_payload.bin
[Atmosphere with emunand]
fss0=atmosphere/fusee-secondary.bin
[Atmosphere over sysnand]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
[OFW - Stock]
stock=1
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin

CTCaer, 2020-01-16:
You sure that this is the config that is being used?

The function that is called and its arguments are:
https://github.com/CTCaer/hekate/blob/4e5ded7cb3bc29b40e62547014d94b05725eceec/bootloader/hos/fss.c#L82

`h_cfg.updater2p` ? `true` : `false`

The functions basically takes a `force` argument:
https://github.com/CTCaer/hekate/blob/4e5ded7cb3bc29b40e62547014d94b05725eceec/bootloader/main.c#L255

And this argument is set by the value ofh_cfg.updater2p, which is `updater2p=` from the [config] in the hekate_ipl.ini. 

The default if missing, is 0 (false):
https://github.com/CTCaer/hekate/blob/4e5ded7cb3bc29b40e62547014d94b05725eceec/bootloader/config/config.c#L46

ELY3M, 2020-01-16:
Thank you!  I will set that in my config and see if it helps.

ELY3M, 2020-01-16:
this is not working for me.  the reboot_payload.bin still getting overwritten.  

[Atmosphere Test]
payload=atmosphere/reboot_payload.bin
[Atmosphere with emunand]
updater2p=0
fss0=atmosphere/fusee-secondary.bin
[Atmosphere over sysnand]
updater2p=0
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
[OFW - Stock]
updater2p=0
stock=1
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin

CTCaer, 2020-01-16:
That's not what I meant. `updater2p=` is not a [whatever] boot section key.
It's a `[config]` section key.

But as I said, the default it's off.
Anyway, I'll recheck the code tomorrow.

CTCaer, 2020-01-17:
Thanks you for reporting that.

It was actually a bug. Fixed in https://github.com/CTCaer/hekate/commit/91759aba95434991955b407470fe426bd40bdc9b

ELY3M, 2020-01-20:
Thank you,  when I compile the latest commit, I got hekate.bin in output folder.    but when I boot the switch with the hekate.bin that I compiled.  it just gone to black screen....   I already updated the .bs0 files in the sys folder.  is there something I missed when I compile?   it compiled perfect.

ELY3M, 2020-01-20:
I also tested it on my own switch. it resulted in a black screen.

CTCaer, 2020-01-20:
Currently master and dev are brokem because the payload is over limits. 
You need to comment out a tool or two in hekate main.

For example:
[image]

ELY3M, 2020-01-20:
Thank you,  now my compiled payload is working now.

CTCaer, 2020-01-20:
Always look that the reported values. It says `Payload size is` and also the max which is 126296 bytes.

Unfortunately I didn't check that when I made the commits and now both master and dev are "broken".

CTCaer, 2020-03-03:
Pushed to a release

closed 2020-03-03 by CTCaer