CTCaer/hekate #346: gray or black screen

Roozbenz, 2020-01-20:
hello im new with this stuff and im not good at eng so sorry about that.
im using latest version of hekate 5.1.1 after i hit the fusee-primary.bin on hekate screen nothing happend just console turns black and turns on with gray screen nothing happens . 
what should i do ? 

sorry im newb

Roozbenz, 2020-01-21:
anyone? :D im little nervous about this

CTCaer, 2020-01-21:
Do you see the display and backlight turn off and then turn on when you launch it?

Roozbenz, 2020-01-21:
after i hit the fuse screen turns black for maybe 2 sec , after that screen turns on with gray screen and nothing happens . no splash screen nothing
after soft reset console turns on normaly 
for the record i hacked once a mounth ago and i did somthing stupid then i reset factory the console and this time this happend to me

CTCaer, 2020-01-21:
The moment you see the screen turning black after hitting it, it already left hekate and it's an issue with fusee-primary.bin (or any payload used).

It's not an issue with hekate, especially when hekate is not used to load the atmosphere files natively.
Can't help you with that.

closed 2020-01-21 by CTCaer