CTCaer/hekate #347: "Unknown pkg1 Version. Unable to launch HOS!"

ameturpoet, 2020-01-22:
I just used this tool (https://github.com/Gabri3lZ/SwitchrootAndroidUtils) to format my new SD card to triple-boot sysNAND, emuNAND, and SwitchAndroid, but now when I'm trying to boot into emuNAND CFW, Kosmos gives me the error "Unknown pkg1 version. Unable to launch HOS!"

I know I have the latest hekate and atmosphere, and I'm on 9.1.0. This worked on my previous SD card, so I'm not sure what the issue is. Switchroot Android boots up fine, but not the actual Switch OS.

CTCaer, 2020-01-22:
You copied over the emuMMC partition?
If yes, then you forgot to change the offset in emummc.ini.
Just run emuMMC migration in hekate. It will find it and set it.

closed 2020-01-22 by CTCaer

ameturpoet, 2020-01-23:
> 
> 
> You copied over the emuMMC partition?
> If yes, then you forgot to change the offset in emummc.ini.
> Just run emuMMC migration in hekate. It will find it and set it.

Hekate said that there was no emuMMC partition or file available when I tried that, despite it existing in the emuMMC directory. I'm trying to recreate a file-based emuMMC right now.

CTCaer, 2020-01-23:
That tool only creates the partition btw.
You still need to run Create emuMMC (partition based) in hekate.
As I said, if you copied it from the old one, it would have been ok.

You skipped answering that question. It was very specific and that's why I continued with `If yes`.

So now I automatically assume that you never copied it or created it. The `file based` you created will work but it's extremely slow currently.

majuss, 2020-01-25:
@CTCaer I resized my SD card and put the emuMMC partition at the end. So I'am also unable to boot right now and hekate can't find the emuMMC when I try to migrate :(
Is there a way to find the boot0 manually? I'am able to operate Linux :)

CTCaer, 2020-01-25:
Hekate can't find the emuMMC because there's nothing to be found. Creating the partition is not enough. It's just an empty partition.
You should also create the emuMMC there or copy the old one.

XxFlamexTrollxX, 2020-07-29:
So when I tapped on launch and hit atmosphere fssoEmuMMC, the hekate system said there was a "unknown pkg 1 version." how would I correct the error?