CTCaer/hekate #348: hekate v5.1.1 SD card and emmc data (@lba 00000000) do not match. Verification failed.

hoonsubin, 2020-01-22:
Using a legit SD card, tried multiple cards formatted as exFAT 128k. One is 128GB and another is 256GB (cost me a fortune).
The verification would always fail at 100% and nothing below that.

CTCaer, 2020-01-23:
I want model names and symbols of the cards (plus model number. For example Samsung Evo+ 256GB is MB-MC256G).

And also a screenshot (tap 3 fingers) or a photo of `Console info` -> `Sd card` per card.

hoonsubin, 2020-01-23:
@CTCaer I'm so sorry for the lack of details in the post.
[image]
The progress bar would reset to 0% after it hits 100%.
[image]
[image]

SanDisk Ultra SDSQUNS-128G-GN6MN
SanDisk 256GB Ultra MicroSDXC UHS-I

only exFAT 128KiB cluster will cause this issue. Using a FAT32 32KiB will display an error at around 4% (I assume due to the size limit of a single file in FAT32).

 If there's any other information required, please tell me.

CTCaer, 2020-01-23:
Yes, the cards appear legit.
**What concerns me is the 191 minutes on a empty SD card.**
Even though it's a class 10 only (no U1/U3, so 10MB/s), it should not get this speed. exFAT or not.
This looks like a class 4 sd card. But your cards' info is too good and extensive to be fake.
**You have similar times with the 256GB card on exfat?**

So based on the above (legit cards):
Normally such slow speeds can be produced by hekate trying to write the same sectors over and over because of I/O failure. (It does 30 retries).
If you are switching/ejecting the sd cards constantly, it might be an issue with your sd card reader connector to main board pins.

Lastly, FAT32 failing at 4%, it failed from I/O issue for sure and not at verification (4% is ~1193GB).
When you have FAT32 it splits the file, so it's not an issue of file size. And verification happens at each part. So the first verification is at 6.8%.
**I need you to tell me if SD or eMMC failed at that point with fat32.**

**Btw, what tool/OS you use to format them?**

hoonsubin, 2020-01-23:
@CTCaer Sorry for the late reply, I've tried everything I can but after some point, I'm getting "failed to init emmc" error and cannot even start a backup.
This is very odd... I'll give it a deeper look and update you when I find something new. I hope this isn't permanent.
FYI Boot 0/1 backups successfully and for some reason, emuemmc can be made from the NAND without any issue, even if it took around 3 hours (till the "failed to init emmc" error that is).

CTCaer, 2020-01-23:
emuMMC creation does not do a verification. That's why it continues.

Anyway, so unfortunately, your issue is a hardware issue with switch's sd reader.
Try a replacement and check the pins in the sd reader and at the connector on top of the motherboard.

closed 2020-01-24 by hoonsubin

vzjxif, 2021-03-17:
@hoonsubin i have a same issue, are you solve it

[image]
[image]

CTCaer, 2021-03-17:
> @hoonsubin i have a same issue, are you solve it
> 
Well sure exFAT support clusters up to 32MB. That doesn't mean that you should use them though...
4MB cluster is so extreme that you are actually decimating performance instead of improving it. Plus that you kill compatibility with embedded platforms that do not expect to have such huge buffers.

vzjxif, 2021-03-18:
> Well sure exFAT support clusters up to 32MB. That doesn't mean that you should use them though...
> 4MB cluster is so extreme that you are actually decimating performance instead of improving it. Plus [...]

> Well sure exFAT support clusters up to 32MB. That doesn't mean that you should use them though...
> 4MB cluster is so extreme that you are actually decimating performance instead of improving it. Plus [...]

thx, i format it, it works