CTCaer/hekate #349: emuMMC restore issue

DankAxion, 2020-01-31:
I clicked migrate emuMMC on my 128gb sd card, then I copied all the files to my pc in a folder. I grabbed my 256gb sd card and used tegraexplorer to format it and have a partition ready like I did for my 128gb. Then I transferred all the files I had on my pc from the 128gb including the emuMMC folder to the 256gb and I launched Hekate and clicked migrate emuMMC again. I launched the CFW and now all my installed games are missing. Is there a way to recover my games and progress? I did a partition emummc btw.

m4xw, 2020-01-31:
Hm?
Saves are on the emuMMC image, even if your Nintendo folder is gone, they will still be there.
Did you get the "another' consoles data are found" warning on first start?
Did you set a folder for the Nintendo path outside of the emuMMC folder?
Please show us your INI here, hekate as well as emuMMC/emuMMC.ini

DankAxion, 2020-01-31:
[image]

Here is the folder where I copied all the stuff from my old SD card after I clicked migrate emummc. You can see there is a emummc folder there that is 17.7GB in size. I transferred all these files to my new 256gb after I formatted it and put a partition on it with tegraexplorer. I then inserted the SD into my Switch and booted up hekate, I clicked on migrate emummc again like it says to do even with backups and selected SD raw1 on change emummc, it was already enabled for some reason too with the checkmark before I even changed it but just to be sure I selected it again. 

When I booted up my switch none of my games are there but the emummc folder is still 17.7GB so I dont know how to restore it properly, I returned the old sd card so theres nothing I can do with it. 
[image]

Thanks for the quick response and help!

DankAxion, 2020-01-31:
Also I just tried to install a game via tinfoil and nut usb from my pc, but when I click install nut says "cannot create file when file already exists". So I believe my games are still here yet they dont show up in home menu.

m4xw, 2020-01-31:
...

CTCaer, 2020-01-31:
First: You never copied the actual emuMMC image
Second: You always need to run the 2 archive bit options after copying an sd card
Third and most important: Piracy

closed 2020-01-31 by CTCaer