CTCaer/hekate #350: Error while executing payload

estelavg, 2020-01-31:
What I can do? 
[image]

deadair3210, 2020-01-31:
Literally tells you on the screen what to do.....

estelavg, 2020-01-31:
No, I done this but it don't help. I used 4 differents sd cards and in the last I could do it. I don't know why the others didn't work.

CTCaer, 2020-01-31:
what @deadair3210 said.

Dl the hekate release zip (not the source code) and throw the `bootloader` **folder** to your sd card root.

hekate release doesn't replace your hekate configs and payloads, so you can merge it with existing bootloader folder and hit the replace button.

EDIT:
If you have it working on 1 sd card only, it means that the other cards are fake or have improper/corrupted filesystem or your sd reader is busted.
But I don't see an error message from the mounting code, which is strange

estelavg, 2020-01-31:
I has downloaded the release zip and the message didn't change and the other sd cards are not corrumpted because I use on my 3DS and it works.

PS: The switch is new and I'm doing a clean installation.

CTCaer, 2020-01-31:
not corrupted means improper filesystem then.
Init them so it creates the MBR and create a partition and format it after.
With a proper tool

azoreseuropa, 2020-02-09:
> I has downloaded the release zip and the message didn't change and the other sd cards are not corrumpted because I use on my 3DS and it works.
> 
> PS: The switch is new and I'm doing a clean installation.

I had the same problem as you. I resolved it on my own. Please let me know so I can help you. Very easy.

closed 2020-02-29 by CTCaer