CTCaer/hekate #351: Waarning: Nyx is missing!

Myselfs-s, 2020-01-31:
I formated my Samsung sd card and copied bootloaded carpet in it, it says nyx is missing and when i try to launch hekate it says failed to initiate sd card

CTCaer, 2020-01-31:
A missing bootloader carpet means that the format you did messed with MBR.
Either it initialized wrongly or it converted it to GPT. You need to init the sd card with a proper tool and then format it.

This because I assume it worked before formatting it.

closed 2020-02-29 by CTCaer

HeroCC, 2020-05-13:
Something for future folks: make sure your SD card is in the switch, not your laptop :sweat_smile:

rock20xx, 2020-12-08:
I'm actually having the same issue.. it say I need an updated payload but its the most recent.