CTCaer/hekate #352: Updating Hekate from 5.0 to 5.1 but it still says 5.0 on launch menu

combluotion, 2020-02-02:
After Updating the files following the nh-server github steps I can see that my Hekate doesn't update to the new version and results in my switch going on black screen after selecting payload. Maybe there is some configuration replacing it as I've seen in another open issue? I'm not really that techie so I can't be sure...

CTCaer, 2020-02-03:
The procedure is simple:
1. Download latest `hekate_ctcaer_X.X.X_Nyx_X.X.X.zip` from https://github.com/CTCaer/hekate/releases
2. Take the `bootloader` folder from the zip and throw it inside the sd card root, without deleting the old folder.
3. Hit replace files when asked by your file manager
4. Inject the included payload from inside that zip

combluotion, 2020-02-03:
What do you mean by Inject?
I guess just replacing the reboot_payload.bin inside the atmosphere folder?
If that's so, that's what I've been doing with no success...

combluotion, 2020-02-03:
Just understood I had to use the atmosphere-injector to inject hetake into the usb-c thing. Sadly it didn't fix the black screen issue...

closed 2020-02-03 by combluotion

CTCaer, 2020-02-04:
when does that black screen happen?

combluotion, 2020-02-04:
After I select the payload in the Hekate Payload list. It then shows atmosphere images and just after one of the logo that looks like an "A" hangs in a black screen.

CTCaer, 2020-02-04:
Yeah that's not an issue with hekate. The moment the display turns off after you select a payload, it already left hekate.