CTCaer/hekate #354: Kernels problem

Stev3nM, 2020-02-28:
What should I do 
[image]

CTCaer, 2020-02-29:
This can happen if there was sth wrong with decrypting it or hashing it. Either by a corrupted file or SE error. You can reliably produce that?

Stev3nM, 2020-02-29:
im noob for this , anything  can do decrypting or hashing and how to dump pkg 2file. use hakate or sth

Stev3nM, 2020-02-29:
> Try to dump pkg2 (Tools -> Dump pk1/2) and see what it outputs.

it said 
Found pkg1 ('20191021113848')
setp will run to get the keys.
Then rerun this option.

and nothings happened   awalys black screen.
can not return to it 

and other thing when i run hakate or other cfw 
and aways stuck in hakate logo
and its said 
Failed to identify kernel /.
Failed to lauch HOS /
press any key..

CTCaer, 2020-02-29:
Try to update `sept folder` from an official and latest atmosphere release zip.
And also drag n drop the `bootloader folder` again (and hit merge and replace files when you are asked) from the hekate v5.1.1 zip.

CTCaer, 2020-03-07:
I assume you fixed the issue.

closed 2020-03-07 by CTCaer