CTCaer/hekate #355: v5.1.1 freezes when try make eMMC SYS backup from 9.1.0

Ninoh-FOX, 2020-03-02:
when the console go to make a copy of the partition BCPKG2-4-SafeMode-sub this is freezes.

CTCaer, 2020-03-03:
This is probably related to https://github.com/CTCaer/hekate/issues/338

Please try with the new release because I fixed some heap/list issues.

CTCaer, 2020-03-07:
Seems that this issue was resolved (#338). Try again with v5.1.2 / 0.8.5.

Ninoh-FOX, 2020-03-07:
same error, only work if you do a new backup, overwritter not work.

CTCaer, 2020-03-07:
Ah that's what you meant. You are supposed to press the POWER button. It says so.

Ninoh-FOX, 2020-03-07:
Yes, I press power, but when go to make the BCPKG2-4-SafeMode-sub  backup, the screen go to back and the console is freeze.

CTCaer, 2020-03-07:
Yeah just checked it. I can confirm.

CTCaer, 2020-03-09:
Fixed with https://github.com/CTCaer/hekate/commit/8d5c52f087584d8f0c43fca2375abc49ed2da420
Relevant issues: #355, #362 and #338.

Because this regression affects way more stuff than it seems, I will do a critical hotfix release in the next hours-days.

Issue will be closed only when the release is ready.

closed 2020-03-21 by CTCaer