CTCaer/hekate #356: feature request : random bootlogo

Heartbeat-Heartbreak, 2020-03-02:
I made a lot of bootlogos and I'm tired of using only one.
a little random, a lot of fun : )

fennectech, 2020-03-02:
I second this owo

ELY3M, 2020-03-20:
this is easy to do.    add randombootlogos in config and add randomizer in the code.   set /bootloader/bootlogos as the path for bootlogos.

ELY3M, 2020-03-20:
this is not done yet but I can do it in my hekate fork.

wiiNinja, 2021-04-24:
I vote for a selectable logo, not a random one. :)

CTCaer, 2021-04-24:
@wiiNinja 
> I vote for a selectable logo, not a random one. :)

It's supported since ever..
Use `bootloader/bootlogo.bmp` if you want a global one.
Use `logopath=path/to/logo.bmp` in a boot entry if you only want that entry to have a custom bootlogo.

CTCaer, 2021-08-13:
This will not be implemented for size constrain reasons

closed 2021-08-13 by CTCaer