CTCaer/hekate #359: how to hide a boot entry (present in "/bootloader/hekate_ipl.ini")

t33st33r, 2020-03-04:
I want to hide a boot entry without removing it.
Can you add something like "enabled = 0" (1 as default) to allow this?
Another way to do this could be allowing multiline commenting.

Please, excuse my bad english. Thanks in advance.

m4xw, 2020-03-04:
You can just comment it.

CTCaer, 2020-03-05:
Even though I find no reason, there's a trick you can do.
You just comment the section part.
```ini
#[label]
key1=0
key2=1
```

This will attach section `label` to a comment type and then keys will fail to get attached to anything.
(hekate checks if current type is section)

t33st33r, 2020-03-05:
> Even though I find no reason, there's a trick you can do.
> You just comment the section part.
> 
> ```ini
> #[label]
> key1=0
> key2=1
> ```
> 
> This will attach section `label` to a comment type and then keys will fail to get attached to anything.
> (hekate checks if current type is section)

I don't understand...
How the default INI contents will be after these changes?

CTCaer, 2020-03-05:
Lets say your ini looks like this:
```ini
[config]
autoboot=4
autoboot_list=0
bootwait=3
verification=1
backlight=100
autohosoff=0
autonogc=1
updater2p=1

[Stock]
fss0=atmosphere/fusee-secondary.bin
stock=1

[Atmo]
fss0=atmosphere/fusee-secondary.bin
logopath=bootloader/res/bootlogo_atmo.bmp
icon=bootloader/res/icon_atmo.bmp
```
So with that, it shows 2 entries inside hekate: `Stock`, `Atmo`.

If you do what I suggested you will get this:
```ini
[config]
autoboot=4
autoboot_list=0
bootwait=3
verification=1
backlight=100
autohosoff=0
autonogc=1

[Stock]
fss0=atmosphere/fusee-secondary.bin
stock=1

#[Atmo]
fss0=atmosphere/fusee-secondary.bin
logopath=bootloader/res/bootlogo_atmo.bmp
icon=bootloader/res/icon_atmo.bmp

```

Now adding # to `[Atmo]` causes it to go missing from the launch menu and it will only show `Stock`.

closed 2020-03-07 by CTCaer