CTCaer/hekate #361: Logging / Error logs?

Drikanyx, 2020-03-07:
Forgive me if this is a particularly dumb question, but are there logs implemented for troubleshooting issues? The "Logs" button doesn't seem to do anything, seems to turn itself off when I change menus.

CTCaer, 2020-03-07:
That button is for troubleshooting HOS booting issues.
What do you want to troubleshoot?

Drikanyx, 2020-03-07:
Exactly that - hekate's launch button just loops back to Hekate and I can't see any faults in the .ini. manual boots work fine (from payloads/fusee-primary)

CTCaer, 2020-03-07:
Payload chainloading is straight forward and there's 0 logs because it directly calls the payload.
As for the launch method, I will need to see the section that fails.

Drikanyx, 2020-03-07:
[Atmosphere FSS0 EmuMMC]
payload=atmosphere/fusee-primary.bin
kip1=atmosphere/kips/*
icon=bootloader/res/emu_boot.bmp

{}

I've also used fss0=atmosphere/fusee-secondary.bin in place of the payload method. same fault.

attached the file as well, plaintext because filetypes
[hekate_ipl.txt](https://github.com/CTCaer/hekate/files/4301846/hekate_ipl.txt)

structure of bootloader on the SD here:[bootloader.txt](https://github.com/CTCaer/hekate/files/4301845/bootloader.txt)

CTCaer, 2020-03-07:
Why your ini is like this?
Anyway try the below one. It's cleaned, config section at top, removed whitespace before captions and also removed the divider captions (they are only used in TUI and not in GUI).
Also stuff that are for hekate in the payload method, are ignored. It doesn't work like that. Only logopath and icon keys are used.

```ini
[config]
autoboot=0
bootwait=1
verification=1
backlight=100
autohosoff=0
autonogc=1

{------ Atmosphere ------}
[Atmosphere via Fusee]
payload=atmosphere/fusee-primary.bin
icon=bootloader/res/emu_boot.bmp
{}

[Atmosphere FSS0 emuMMC]
fss0=atmosphere/fusee-secondary.bin
kip1=atmosphere/kips/*
icon=bootloader/res/emu_boot.bmp
{}

[Atmosphere FSS0 SYS]
fss0=atmosphere/fusee-secondary.bin
kip1=atmosphere/kips/*
emummc_force_disable=1
icon=bootloader/res/sys_cfw_boot.bmp
{}

[Stock SYS]
fss0=atmosphere/fusee-secondary.bin
stock=1
emummc_force_disable=1
icon=bootloader/res/stock_boot.bmp
{}

```

Also I would suggest to update hekate if you haven't already. Your config section looks old.

Drikanyx, 2020-03-07:
Thanks for your patience with me, that config file worked. Goes to show I'm not as clever as I think I am. I know this isn't supposed to be tech support, so thank you.

The .ini was one provided by a modding guide, edited poorly by myself... and I'm using the latest version, I think the fact it's a borrowed and frankensteined ini is why it looks so old.

The differences between yours and mine aren't hugely different (to me, at least), obviously config at the top and removal of the whitespace and headers - what could have been causing the boot loop, out of curiosity? Poor formatting?

CTCaer, 2020-03-07:
It was caused from the config. The autoboot and the hekate native hos launch are working quite similarly.

The autoboot method tries to match the config section autoboot key with the number in the list.
Nyx->hekate main -> HOS reads the config and then tries to match the special storage's boot cfg.
That storage is basically an external communication method to hekate main.

To sum up, config at top and a newline after it, are mandatory. After that you can add whatever.

closed 2020-03-07 by CTCaer

yeet-dot-org, 2024-01-13:
> That button is for troubleshooting HOS booting issues. What do you want to troubleshoot?

Sorry for a dumb question, but i cant find my the log file or a folder where they would be after enabling logs. im booting to sysmmc CFW by the way

CTCaer, 2024-01-13:
They are on screen. Take a video to look at them