CTCaer/hekate #362: Froze during eMMC backup

justafridge, 2020-03-07:
Hi, I'm was using Hekate 5.1.2 to make a NAND backup using an 8GB microSD (I know it's small) and everything went perfectly until it reached 100% and then froze at 100% while verifying.
The Switch has never run any kind of CFW before and I've just tried biskeydump v9 on it to test fusée gelée, is there any risk in shutting down the Switch or do I plug it in and wait to see if it completes properly?

CTCaer, 2020-03-07:
This is probably related with https://github.com/CTCaer/hekate/issues/355 and partly with https://github.com/CTCaer/hekate/issues/338
I found parts that cause the issue, so I'm working on a fix.

Yes you can force shut down it at that point. hekate flushes the data safely and basically when you see 100% it really finished.

justafridge, 2020-03-07:
ok thanks :)

closed 2020-03-07 by justafridge