CTCaer/hekate #366: error: unknown pkg1 version.

lpt2007, 2020-03-17:
Hi I created my sdcard using this turtorial:
[](https://www.partitionwizard.com/partitionmagic/set-up-emummc-emunand-on-nitendo-switch.html))

Then I go to hekate --> emuMMC --> Create emuMMC --> SD Partition - Part 1 - Cantinue

Creating process takes about 30min and than say finished. Then I press close in right top corner.

Then I enabled emuMMC with Change emuMMC -- SD RAW 1 (screen says The emuMMC configuration was saved to sd card!) -->Ok.

Now If I Close --> Launch ---> CFW (EMUMMC) I get error: unknown pkg1version.

If I try "Migrate emuMMC" before I try first launch cfw I get screen Found SD Partition emuMMC! Do you want to repair the config for it? And when I press continue -->close --> launch --> cfw (emummc) I get same error.

If I go after error to emuMMC --> Migrate emuMMC I get screen No emuMMC or foreign emunand founds!

fennectech, 2020-03-17:
Create a new EMUMMC after wiping the emummc partition     This is sounding like something went wrong in the process of creating your emummc

CTCaer, 2020-03-17:
Take a screenshot or photo of `Console Info -> SD card` and post it here

lpt2007, 2020-03-17:
> Create a new EMUMMC after wiping the emummc partition This is sounding like something went wrong in the process of creating your emummc

I try multiple times.

> Take a screenshot or photo of `Console Info -> SD card` and post it here

[image]

urherenow, 2020-03-17:
I could be wrong, but it looks to me like your card is fake. I’m shown all of the info on the left, that you appear to be missing.

lpt2007, 2020-03-17:
> I could be wrong, but it looks to me like your card is fake. I’m shown all of the info on the left, that you appear to be missing. Is your card _REALLY_ 1TB?
> 
> [image]

Card is realy 1Tb. Not fake.

urherenow, 2020-03-17:
It’s either fake or defective. Sorry. It’s not reporting vendor, serial number, or anything like that. Try a different card.

fennectech, 2020-03-18:
This is DEFINITLY fake.  It lies about capacity https://youtu.be/J-D6tYBX8vE

CTCaer, 2020-03-18:
@lpt2007 
Your card is definitely fake:
[image]

Not only that, but it's a class 4 card even. Probably its real size is 16GB. Maybe less. Hw2testw will tell you.

Have in mind, that even if you partition it to its real size in order to salvage it, it will still be extremely slow.

closed 2020-03-18 by CTCaer

XxFlamexTrollxX, 2020-07-24:
How do I fix the unkown pkg 1 error?

impeeza, 2020-07-24:
In this case:. Using a new SD card. A verified one

XxFlamexTrollxX, 2020-07-25:
then how would I know if the SD is verified?

impeeza, 2020-07-25:
H2testw. Google it

XxFlamexTrollxX, 2020-07-29:
Alright, so I tested my sd card with h2testw. It normal and without error. So what do I do from here to correct my unknown pkg1 error?

impeeza, 2020-07-29:
You could go to discord, wikis or other support channels and read the guides. On Gbatemp there is a post about that error and how to fix it, i think is something about missing files or using the wrong version of firmware update

XxFlamexTrollxX, 2020-07-29:
Alright, thanks!

XxFlamexTrollxX, 2020-07-29:
Yeah going to gbatemp was helpful.