CTCaer/hekate #369: 5.1.2 breaks booting HOS 6.1.0

AuroraWright, 2020-03-21:
Booting the latest vanilla AMS with this in the hekate_ipl.ini leads to the Nintendo splash screen rather than the AMS one, and AMS not loading at all with HOS 6.1.0:
```[Atmosphere FSS0 SYS]
fss0=atmosphere/fusee-secondary.bin
emummc_force_disable=1
icon=bootloader/res/sys_cfw_boot.bmp```
Leaving the bootloader folder intact and using 5.1.1 boots AMS successfully.

AuroraWright, 2020-03-21:
It's related to the sept from fss0 code as reverting 5a9bbaf900fe22b55b8e8d56d9c7f7ed1cc7e21d , 01a767cbfaa2dc6a4a1d3ab55140b299395a3813 and 8584493c7fc7964d7a694b0cd5823ca336a05463 fixes the issue.

CTCaer, 2020-03-21:
Thanks for bisecting.
I somehow removed the crucial check. This forced every < 7.0.0 boot on SYS to be stock.
Fixed in https://github.com/CTCaer/hekate/commit/a8db070755cbb15b1446a6db88e1a5321fd53b8a

The issue will remain open until a new release.

closed 2020-03-21 by CTCaer