CTCaer/hekate #370: stock FW through hekate launch reverted to older console state

grubsong, 2020-03-21:
I haven't really reported issues like this before and I don't have any logs, my apologies, so I'm sorry if this isn't much help.

I've been running CFW on sysnand, and have the newest version of hekate (v5.1.2) and atmosphere (v0.10.5) and I solely use CFW for Pokemon RNG through the tools [capturesight](https://github.com/zaksabeast/CaptureSight/releases) and [luxray](https://github.com/3096/luxray).

 I did my usual RNG stuff, switching between the homebrew menu and the game, etc. I got the frame for the Pokemon I wanted, saved, and hit restart on the switch, which of course, booted Hekate. I wanted to switch to stock fw so I would be using stock fw online when I hosted the Pokemon raid (it's naive, considering my switch is 'tainted' by cfw, but I figured it would help). Hekate booted to stock, but when I opened up Pokemon Shield, it instead had a save file from 2+ hours ago. I'm aware Pokemon Sword/Shield have a backup save system, so I considered that it could have been that, but when I turned the switch fully off, and back on again without CFW, my Pokemon was back, with that 2+ hour progress restored.

I've switched to stock fw with hekate before, but never had this issue, I believe it might be something in the newest update?

CTCaer, 2020-03-21:
Nope, hekate can't affect that. You are probably mixing up something.
You sure you don't use emuMMC or pokemon doesn't do anything timely with that backup save system?

Anyway, savedata is purely a HOS thing and not a bootloader (hekate) thing.

If you still think something is wrong, it's better to make sure *what* you are launching.
(check the actual hekate_ipl.ini and the profiles you boot and also if you have emuMMC and it's not disabled in the ini).
And report that issue to Atmosphere repo.

grubsong, 2020-03-21:
There could be a chance I accidentally hit emuMCC to boot, but... in that case the Pokemon that popped up "wrong" probably shouldn't have been an event one. Thank you anyway, Sorry for the hassle.

closed 2020-03-21 by grubsong