CTCaer/hekate #371: NAND Backup Percentage Issues

OldGnashburg, 2020-03-22:
The NAND backup percentage bar is having issues. It's gone from being stuck at 0 all the way to 6, or it hits 24% and goes back to 20%, when it's having the issues, the percentage bar is a different color.

CTCaer, 2020-03-22:
The percentage is normal. The color of the bar denotes if it's copying or verifying. And ofc for verification, it must start from the beginning of that part.

As for the text, can you take a photo?

OldGnashburg, 2020-03-22:
Thanks for explaining the percentage thing. Also the text issue isn't always there, I used Hekate again shortly after and it was fine. It made me think sending the payload got screwed up. But if you get the issue too than maybe it isn't just me.

CTCaer, 2020-03-22:
Found the issue and it's a lvgl lib bug. It doesn't check some stuff when it sets text label.
I'll change description and mark this as bug.

CTCaer, 2020-03-22:
Fixed in https://github.com/CTCaer/hekate/commit/a39ba2cd71190d3e289722cc7bd1199baf7f3af1

closed 2020-04-13 by OldGnashburg