CTCaer/hekate #372: Iused the panel calibration reverse-forward fuction and touch in HOS dead.

maskmark, 2020-03-22:
now testing to get back.

maskmark, 2020-03-22:
no luck. tested nyx6.bin, still touch in ofw , cfw(nand/emummc) all gone.
I should not do that since I don't have any issue.

maskmark, 2020-03-22:
G, I even Re-initialed the console, still no luck.
guys , never use this if you don't have a touch funtion.
NOW in all cases it's like the wrong touch calibration in the old NYX for me now.
sad.

maskmark, 2020-03-22:
so, where does the console store the calibration info? in the touch IC?
wonder if a tool version of this forward fuction would resolve my problem,since I even don't have a right calibration info in my console/HOS now.

maskmark, 2020-03-22:
it's a bit Dangerous to use the funtion in some cases ( like if you have a touch issue in NYX before).
be careful to use this.

maskmark, 2020-03-22:
I tested the nyx 1-6 test version
except 5,  it even not give a 12345 screen but a normal nyx  now(but no touch of course).
what changed?confused.

CTCaer, 2020-03-22:
This is not a chat to spam. Just edit your message.
The tool is not supposed to be used if you don't have issues btw.

So lets get it from start.
- Why you run the function? Nyx or HOS issue?
- What the touch screen test shows inside HOS? The one with the lines.
- Were you touching the screen when you were pressing VOL buttons?

maskmark, 2020-03-22:
> This is not a chat to spam. Just edit your message.
> The tool is not supposed to be used if you don't have issues btw.
> 
> So lets get it from start.
> 
> * Why you run the function? Nyx or HOS issue?
> * What the touch screen test shows inside HOS? The one with the lines.
> * Were you touching the screen when you were pressing VOL buttons?
> 
> EDIT:
> Also stop doing random stuff like initializing the console or try random binaries.
> The IC stores the calibration in its non volatile storage and it can be easily manipulated.

Sorry for the confusing discription.

1. I ran the new function only for test "forcing a calibration". there was no issue since I used the test version nyx6.zip u gave me(before that I had a touch issue in nyx).

2.in HOS(including stock , nand cfw and emummc cfw) the touch screen acts just like it was in NYX.only the narrow top line can be touched.it was like a wrong calibration info "forwarded" to the HOS or touch IC.

3.I don't think i touch the screen when presssing the VOL buttons.

sorry CTCaer, I just wanted to warn dumb guys like me myself not to do that if no issue occurs.

CTCaer, 2020-03-22:
I can help on this matter but will need discord.
You can use a whatever free vpn to access it.

maskmark, 2020-03-22:
> I can help on this matter but will need discord.
> You can use a whatever free vpn to access it.

believe it or not, they also managed to BLOCK my vpn. will contact you when I could. thanks again for all you did for the scene.

CTCaer, 2020-03-22:
For now, run the calib again. Hold both VOL buttons, do not touch the screen, and inject. After you see nyx homescreen, test the touch. (This must be done with 0.8.6 nyx ofc)

maskmark, 2020-03-22:
> For now, run the calib again. Hold both VOL buttons, do not touch the screen, and inject. After you see nyx homescreen, test the touch. (This must be done with 0.8.6 nyx ofc)

tested. no luck ,it acts like it was in old nyx(no touch in nyx except the top bar).

CTCaer, 2020-03-22:
Doesn't matter where you test it.
The only thing that changes is that if you initialize HOS (sys or emuMMC), it will run the same calibration as in Nyx.

CTCaer, 2020-03-22:
And as I said, you need to find a free VPN solution. Debugging this in here is not simple and too much time consuming.

If you manage to access discord, you will understand what I mean.

maskmark, 2020-03-22:
> Doesn't matter where you test it.
> The only thing that changes is that if you initialize HOS (sys or emuMMC), it will run the same calibration as in Nyx.

well, I realize  ofc means ”of course“ now...
what I did today is to run the fuction and find touch all gone in nyx/ofw/syscfw/emuMMC, 
and the I initialized the console in stock HOS.
till now the touch screen acts like it did in the old NYX. re-calibration did not help me. sorry for the spam info I made.

maskmark, 2020-03-22:
> And as I said, you need to find a free VPN solution. Debugging this in here is not simple and too much time consuming.
> 
> If you manage to access discord, you will understand what I mean.

Okay ,will find you on discord when I get the access. thanks!

CTCaer, 2020-03-24:
Latest tests point to a hw failure that produces random huge sense/force levels to be output to calibration data. So when the MS/SS tuning is done and saved, it has large values and effectively killing touch.

Running ITO production test, points to either Sense channel 4 pin being on the air or Force channel 4 pin being shorted to ground.

It's not completely clear which case, because drivers point to ITO_SENSE_OPEN/ITO_FORCE_GROUND as being error 2/3 or 1/2. In this case the error has code 2.

Probably the ITO check will be incorporated to the forced re calibration tool, to mitigate the touch IC allowing such panels to be calibrated.

Lastly, the calibration done is 1:1 with the one done in HOS. This means that if an updated HOS version brings a new touch fw version, it will do a re calibration. This explains why some users have suddenly working or non-working touch with such an update.

CTCaer, 2020-03-24:
This mitigates that issue: https://github.com/CTCaer/hekate/commit/dc9c7fd95c86884d591421da01081ce1bd761150

Can't vouch for HOS not doing it though.

cristianrj9, 2020-03-31:
hola CTCaer como puedo calibrar el indicador de bateria de mi switch nunca hice uso de linux o android 
simplemente sucedio de la noche a la mañana 
la duracion de mi bateria es normal el problema es el indicador ya que cuando cargo al 100% y lo apago para volver a encenderlo despues de unas horas del 100% la consola enciende en 1% pero la duracion de bateria es normal  solo que el indicador no es el correcto como podria calibrarlo con hekate ayuda por favor

CTCaer, 2020-04-15:
So a small write up about this issue:

@maskmark 's touch panel has a HW issue on the the Sense channel 4 pin. The issue is that this channel is not connected to the touch IC.

The touch IC fw has a bug where it allows touch panel calibration to run even though there's a HW issue.
This causes the calibration sequence to fail and put random huge values in its tuning. Effectively killing touch by zeroing the sensitivity.

Another issue on that, is that HOS can also trigger the calibration when it includes a new FW version for the touch IC. Effectively killing the defective touch panel.
This is easily demonstrated by erasing touch IC fw and then booting HOS. HOS will then see that and update the fw and then do a calibration.

Only fix to that, while keeping the defective touch panel, is to write manually a new calibration from a known working one.
Unfortunately, due to the closed nature of FingerTip S products and that their FW and ROM is always customized for each customer (here Nintendo), these addresses are unknown.

Calibrating changes values in SRAM, but what we need is a way to read the EEPROM and validate addresses. The command that does that is not known.

Lastly, as said before, this was mitigated in Nyx and it now does not allow the calibration to run if it finds that the panel has a HW issue. But the problem that HOS can do the calibration remains...

CTCaer, 2020-05-01:
New version has the calibration tool and also checks HW first

closed 2020-05-01 by CTCaer