CTCaer/hekate #373: can I backup a fake SD?

Ooskk, 2020-03-22:
I bought an SD and verified through H2testw. I guess it's a little over 380 reports as 500.

Error writing file 'G:\379.h2w', offset 0x1c900000.
It is still possible to verify the test data written up to this point.
(The wrong diskette is in the drive.
Insert  Code 34)
Warning: Only 482004 of 482070 MByte tested.
Writing speed: 20.1 MByte/s
H2testw v1.4

Is there a possible way to still do backups?

maskmark, 2020-03-22:
if u ever wanna use this backup, no.
never.

Ooskk, 2020-03-22:
are you implying there is an alternative?

CTCaer, 2020-03-22:
You can try, nyx verifies the backups. On the other hand, even if it succeeds, a small change after it, might corrupt it.

Ooskk, 2020-03-22:
Curious. So you think it can randomly overwrite even before reaching capacity? I thought I could be safe since I can just keep actual files under 250 GB or so.

CTCaer, 2020-03-22:
You need to change partition size for that to work. FAT keeps backup tables at the end of the partition also.

closed 2020-03-23 by CTCaer