CTCaer/hekate #374: Black screen after injecting the hekate payload to my switch.

amitbartfeld, 2020-03-23:
when I'm injecting payload into my switch it says that it's ok but the screen stays black (trying to use hekate).

I'm at 9.2.0 and my serial num is XAJ7004...

I had CFW on my switch before and I tried to update it to 9.2.0 and from then it's not working.

It's not working with tegra + with an external RCMloader.

urherenow, 2020-03-23:
Define “external RCMloader”. You need to, at a minimum, extract the entire contents of the latest Atmosphere (all folders and overwriting all files), same with Hekate, MINUS the hekate.bin, and hekate.bin is what you inject. If you have a dongle of sorts that does the injection, you need to rename hekate.bin to payload.bin, and get it onto your dongle.

amitbartfeld, 2020-03-23:
I did all the following except the lest line... (didn't understand what do you mean in it. can you explain?)

CTCaer, 2020-03-23:
Did you drag n drop the latest `bootloader` folder?
Also does your switch have enough battery?

amitbartfeld, 2020-03-23:
yes to both.

urherenow, 2020-03-23:
You plugged your rcmloader to your pc (via microUSB) deleted payload.bin from the top folder, dropped hekate.bin in there, renamed it to payload.bin, and also made sure the rcmloader was set to the correct folder before plugging it in to your switch? (Blue light, iirc...)

amitbartfeld, 2020-03-23:
yes I did

CTCaer, 2020-03-23:
Try running this and tell me what it gets printed on the screen:
[check.zip](https://github.com/CTCaer/hekate/files/4371984/check.zip)

amitbartfeld, 2020-03-24:
Ok Thanks

amitbartfeld, 2020-03-24:
The TegraRcmGUI is crashing.
and with the RCMLoader The screen stays black...
:/

CTCaer, 2020-03-24:
Yeah the issue is not hekate then.

This binary is a simple hw init (without sd stuff or extra, no bpmp freq change or dram freq).
If it doesn't work, the issue is in your cable or loader or your usb port, or VOL/Jig buttons.

amitbartfeld, 2020-03-24:
Ok thank you very much!

closed 2020-03-24 by amitbartfeld