CTCaer/hekate #378: Hekate 5.1.3 and 5.1.2 huge boot time

Patrick5897, 2020-04-08:
Since my switch is on 9.2.0, I experience a huge boot time. It takes about 12 seconds before the sept screen appears. I tried to change the auto boot time and for one reboot after the change it boots in like three sceonds. At the second reboot it again takes about 12 seconds.

CTCaer, 2020-04-08:
Go to console info -> sd card and take a screen shot (3 finger tap) or photo

Patrick5897, 2020-04-08:
[Screenshot](https://imgur.com/OULP2Qe)

CTCaer, 2020-04-08:
What is the size of your reboot_payload.bin in atmosphere folder?

Patrick5897, 2020-04-08:
> 
> 
> What is the size of your reboot_payload.bin in atmosphere folder?

123 KB

CTCaer, 2020-04-08:
Well this weird.
What are the contents of hekate_ipl.ini?

Also are you using /bootloader/update.bin?
If yes, upload it here, together with the reboot_payload.bin (zip them) so I can examine their headers.

Patrick5897, 2020-04-08:
My hekate_ipl.ini:
[config]
autoboot=1
autoboot_list=0
bootwait=3
verification=1
backlight=100
autohosoff=0
autonogc=1
updater2p=0

{AtlasNX/Kosmos v13.2}
{}
{Discord: discord.teamatlasnx.com}
{Github: git.teamatlasnx.com}
{Patreon: patreon.teamatlasnx.com}
{Pegascape DNS: pegascape.sdsetup.com}
{}

{--- Custom Firmware ---}
[Atmosphere (EMUMMC)]
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_atmosphere.bmp
{}

I don't use update.bin.

CTCaer, 2020-04-08:
Well something's wrong with the files in your sd. 
Drag n drop the whole contents of atmosphere release zip and drag n drop the bootloader folder from hekate release. When asked to merge folders and replace files, hit yes.

Patrick5897, 2020-04-08:
Thanks it works now after clean start :)

CTCaer, 2020-04-08:
I wonder what was corrupted.
Anyway closing, as `corruption fixed by refreshing files`.

closed 2020-04-08 by CTCaer