CTCaer/hekate #380: Not booting past the Switch logo

ThatAnarchyDude, 2020-04-15:
I replaced all existing Atmosphere files with the new ones Scires posted on Twitter today.
I replaced all of the existing Hekate files with the new ones CTCaer posted on Twitter today.
I attempted to boot the CFW (firmware boot in between EmuNAND & Stock on the menu), and I was got to the Switch logo before being presented with this.
[image]

Would my error logs help?
Crash Reports:
[01586912723_0100000000000014.log](https://github.com/CTCaer/hekate/files/4478692/01586912723_0100000000000014.log)
[01586912724_0100000000000014.log](https://github.com/CTCaer/hekate/files/4478693/01586912724_0100000000000014.log)
[01586912860_0100000000000014.log](https://github.com/CTCaer/hekate/files/4478694/01586912860_0100000000000014.log)

Fatal Reports:
[01584215621_0100000000001007.log](https://github.com/CTCaer/hekate/files/4478697/01584215621_0100000000001007.log)
[01586911481_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478698/01586911481_0100000000001000.log)
[01586911509_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478699/01586911509_0100000000001000.log)
[01586912548_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478700/01586912548_0100000000001000.log)
[01586912708_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478701/01586912708_0100000000001000.log)
[01586912782_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478702/01586912782_0100000000001000.log)
[01586913846_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478703/01586913846_0100000000001000.log)
[01586914015_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478704/01586914015_0100000000001000.log)
[01586914164_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478705/01586914164_0100000000001000.log)
[01586914617_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478706/01586914617_0100000000001000.log)
[01586914657_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478707/01586914657_0100000000001000.log)
[01586914690_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478708/01586914690_0100000000001000.log)
[01586914727_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478709/01586914727_0100000000001000.log)
[01586914754_0100000000001000.log](https://github.com/CTCaer/hekate/files/4478710/01586914754_0100000000001000.log)

I can supply the Crash/Fatal dumps as well, if that would help.
I've never had to post an issue with Atmosphere or Hekate before...so I'm not really sure where to go from here or what else I can do to help.
Thank you.

masagrator, 2020-04-15:
Have you deleted custom theme?

SquirrelKiev, 2020-04-15:
Yeah. Looks like a custom theme problem. Try deleting atmosphere/contents/0100000000001000/ and see if that fixes it.

CTCaer, 2020-04-15:
Errors on qlaunch are normally because of old themes. Also this is not a hekate issue

closed 2020-04-15 by CTCaer