CTCaer/hekate #381: ACNH Crash Issue

Orvurator, 2020-04-15:
So I was save editing, and after a restore it said my save data was corrupted. I thought this was an issue with something I programmed in wrong so I proceeded to change revert to an older save, and the same thing occurred. I then deleted the save data and after ten tries of opening the game it froze on the characters talking. I then redownloaded it and it won't even open the game.

Orvurator, 2020-04-15:
This is for Animal Crossing btw

Conan179, 2020-04-15:
Why do you think Hekate is the problem?

Orvurator, 2020-04-15:
haha I apologize wrong thread. Sorry to disturb. I have talked to some others who have had issues with it and they say it's a common problem that is being worked on.

closed 2020-04-15 by Orvurator