CTCaer/hekate #383: Error screen after updating to latest Atmosphere and Hekate

idmew, 2020-04-15:
Hi, I just updated to latest releases to Atmosphere 0.11.1 and Hekate 5.1.4 and when I try to open Album I get an error and I have to restart my Switch.

My games are running fine, I just can't access the homebrew menu.

My firmware is 9.2 and I'm not running any custom themes.

Any idea how to fix this issue?
[image]

CTCaer, 2020-04-15:
Run both unset and fix archive bit options  in hekate. They are in tools, second tab

idmew, 2020-04-15:
Hi CTCaer, thank you kindly for the fast reply.
I did as you asked and run both unset and fix archive but it did not solve the issue and I still cant access homebrew menu.

BetaLeaf, 2020-04-15:
I thought that would happen. Real issue is NxThemeInstaller was not uninstalled before updating. Delete the `SD:/atmosphere/contents/0100000000001000` folder.

idmew, 2020-04-15:
Hello BetaLeaf, I do not have NxThemeinstaller installed for the file  "0100000000001000"

BetaLeaf, 2020-04-15:
That Program ID 0100000000001000 suggests that it has something to do with NxThemeInstaller and the theme you have installed. I had to uninstall my theme when I updated too. If you came from a really old version of Atmosphere, it may be in `SD:/atmosphere/titles/0100000000001000`, though if I recall correctly, the latest doesn't load that directory anymore, so it shouldn't matter. 

You may want to try renaming the atmosphere folder to atmosphere.old and placing a fresh copy of the atmosphere folder from the latest release onto your sd card.

CTCaer, 2020-04-15:
Delete /atmosphere folder and drag n drop the latest one. Themes do not only come with 1000 in their name.

BetaLeaf, 2020-04-15:
I'm just going by what is shown in the screenshot provided

CTCaer, 2020-04-15:
And you are right. archive bits have nothing to do with that. I didn't check the ID.
0100000000001000 always denotes theme issues

BetaLeaf, 2020-04-15:
although, isn't he going to want to go ahead and re-run the archive bits since he just wrote a fresh atmosphere copy or will it not matter in this case?

CTCaer, 2020-04-15:
That depends on OS. If the OS puts the archive bit in the folders that were extracted, then yes.
The archive bit options must run again

idmew, 2020-04-15:
Hi, once again than you both for all the help.
So I deleted the atmosphere folder and replaced it with the latest version and run re-run the archive bits,  it did not resolve the issue and now all my games say corrupted data :(

idmew, 2020-04-15:
Any other suggestions? I'm stuck at home with nothing to do and now all my games stopped working.
I did every other updates from Atmosphere and Hekate with no problem so I don't understand why it would suddenly stop working.
My sdcard is exfat, but been using it for almost a year with no problem.

CTCaer, 2020-04-15:
Well it's easy to fix. Don't pirate.

closed 2020-04-15 by CTCaer

idmew, 2020-04-15:
I own all the games that are installed on my system.

BetaLeaf, 2020-04-15:
I'm going to assume that the issue is not piracy related but actually the archive bit on the Nintendo folder (either on root of SD or in emunand) is set incorrectly. You can fix it by running Hekate, going to Tools at the top, selecting Archive bit in the bottom right, and select the "Fix archive bit - Nintendo" button. As you can see, this option specifically deals with fixing corruption on Nintendo Games, and as you messed with this earlier, it is likely the culprit. If this doesn't fix your issue, then please open a new ticket with a screenshot of the new error you are getting when you try to launch the homebrew launcher.

idmew, 2020-04-15:
Hi, this very kind of you for taking the time and still helping me after my thread got closed down. I tried running the Archive bit and it display that it did fix the files but unfortunately I'm still getting the same error when I try to access homebrew through Album.Some suggested that I format my SD card to fat32 so might try that I think.Once again thank you for the help.

BetaLeaf, 2020-04-15:
Yea np, with my prior experiences, I had a strong feeling this wasn't
piracy related. If your sd card is using ExFat, yes you should backup your
sd card and format it with fat32 instead. ExFat driver for switch is known
to cause corruption issues. I'd give that a go. You can also try the other
archive button in hekate. What error code are you getting. Can you upload a
new screenshot? The old one you uploaded isn't relevant anymore as we fixed
that issue by replacing your atmosphere folder in a previous step.

idmew, 2020-04-15:
Hi betaleaf, I already started formating my card so won't be able to tell you the error message but will definitely update you once Im up and running again

idmew, 2020-04-16:
Hello again, just an update after formating to Fat32 and copying everything over, I was able to open the homebrew menu again.Thanks again for the help!On Apr. 15, 2020 5:54 p.m., id_mew@yahoo.com wrote:Hi betaleaf, I already started formating my card so won't be able to tell you the error message but will definitely update you once Im up and running again

Gizzy201, 2022-08-15:
> This helped me! Thank you.