CTCaer/hekate #385: Black screen when creating new sd based partition

mbamber1986, 2020-04-15:
Heya All i have created a sd based emmuc on hekate when its completed and i have chosen the actual partion when i boot it does the ams and sept loading screens but then a black screen after failing to boot the rest of the way

CTCaer, 2020-04-15:
Get me a screenshot or photo of Console info -> sd card in Nyx

mbamber1986, 2020-04-19:
Hello thanks for your response as i explained i have managed to create a sd partition based emummc  however the sd file based one creates but does not  load after the atmosphere arrow logo and just g
[image]
ives a black screen
[image]
[image]
[image]
[image]

Sorry after reading original message seems confusing the issue is with sd file based not partition based

deadair3210, 2020-04-19:
Unless it changed recently, file based emunand is incredibly slow compared to partition based or sysnand, it might still be loading , just veeeeery slowly

masagrator, 2020-04-19:
Yup, I confirm. File based is very slow.

CTCaer, 2020-04-19:
Yep, sd **file** based just takes forever to load. Especially with a U1 card.
You can try a **partition** based or wait around a week.

closed 2020-04-19 by CTCaer