CTCaer/hekate #386: No mandatory warmboot or secmon provided.

fennectech, 2020-04-15:
After updating to the latest hekate and atmosphere last night ( 0.11.0 ) i am running into “no mandatory secmon or warmboot provided”. The issue still happens on 0.11.1. I’m able to boot. By chainloading fusee-primary from hekate but i cant boot atmo directly.  My Google fu has failed me.

CTCaer, 2020-04-15:
Your ini is wrong then. This can only happen if fss0 is malformed or if you use the kips way, you forgot to set `secmon=` and `warmboot=` keys

fennectech, 2020-04-15:
i was commenting out the nosigchk with # instaid of ;   Thats where my problem is.

closed 2020-04-15 by fennectech

skudi, 2021-10-05:
I've just made the same mistake when commenting lines in /bootloader/hekate_ipl.ini thanks for this tip :)

frntn, 2022-05-01:
Hello traveller from the future.
Same issue, same fix : I removed my "#" starting lines from the ini file

Blumpkus, 2022-05-09:
> Your ini is wrong then. This can only happen if fss0 is malformed or if you use the kips way, you forgot to set `secmon=` and `warmboot=` keys

explain this in dummy terms please

fennectech, 2022-05-09:
Your hekate_ipl.ini is fucked or /atmosphere/package3 is corrupted

urherenow, 2024-03-12:
Why are you commenting on an issue that was closed FOUR YEARS AGO? Boot semi-stock or stock, or even to OFW from within nyx, then reboot to CFW. Remove cal0blank=1 if you intend to use fw 17.0.1, and setup dnsmitm.

https://gbatemp.net/threads/quick-one-on-atmosphere-dns-mitm-feature.587581/