CTCaer/hekate #389: Booting Stock not working

ollidiemaus, 2020-04-16:
What's the issue you encountered?
When I try to boot Stock sysNand using hekate 5.1.4 with this config:

```
[OFW Stock]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
stock=1
icon=bootloader/res/icon_switch.bmp
{}
```
First the Sept Splash Screen appears then I get a black screen.
But when I'm booting sysNand with atmosphere everything works just fine.
Also emuMMC is booting fine as always.

How can the issue be reproduced?
I just copy paste the atmosphere, sept and switch folder from release 0.11.1 to the SD Card.
Also I copy bootloader from hekate 5.1.4 to the SD Card.
Then I copy the latest fusee-primary in the atmosphere folder.

System Firmware Version
10.0

quiggy, 2020-04-20:
I can confirm this issue report. Same behaviour on my switch.

NickHu, 2020-04-20:
Seems to be caused by the nogc patch

CTCaer, 2020-04-20:
I can confirm. I always burn my fuses so I didn't notice.
nogc breaks on 10.0.0 semi-stock.
I will change the logic for that. Users will still be able to force it with `kip1patch=nofc`, like what I do (because of burnt fuses) when there's a new gc update.

@ollidiemaus @quiggy you can disable the auto nogc for now.

raphamotta, 2020-04-20:
I can confirm that with the latest version of hekate

ollidiemaus, 2020-04-20:
@CTCaer Okay Nice thanks with auto nogc off its working :) Thanks a lot!

raphamotta, 2020-04-20:
Quick question, if I turn auto nogc off it will burn my cart fuses even if I don't use and cart games??

CTCaer, 2020-04-20:
Well no, but eventually you will just forget a cart inside and burn them.

CTCaer, 2020-04-20:
I will mark this as bug to raise awareness.

CTCaer, 2020-05-01:
New version does not apply auto nogc.
Users that still want it, use `kip1patch=nogc`
But if sys is 10.x.x stock will hang

closed 2020-05-01 by CTCaer