CTCaer/hekate #390: Restoring Nand backup 0% -> 100% -> 0%

sharaku17, 2020-04-16:
Hey guys,
i have a juge problem. My friend asked me to restore his nand Backup on his switch again.
The Switch had a clean Nand backup made about a year ago and was on Firmeware 4.1.0
Everything was backed up, rawnand, boot 0 1, prod info and all other keys.

Now the firmeware was still on 4.1.0 so the problem  shouldnt happen because of the firmeware then.
Now i restored the Boot0/1 files without problem and then started restoring the eMMC part and the restoration goes well until reaching 100%. then it goes back to 0% and just freezes and nothing happens anymore. Im forced to force close the switch and of course now because of the failed restoration the switch wont boot anymore. Makes sense.

I tried to do the process again and the same happened again. During the first time i had the original charger plugged in and during the second attempt i had no charger plugged in. It didnt make a difference.

Im at a loss of words and dont really know how to go on now i really need help guys...

sharaku17, 2020-04-16:
i got it fixed by using a very older version of hekate that i found by searching the other issues here where someone had a rather similiar problem while only while doing the backup. with the older version of hekate the restoration didnt freeze and it worked. Switch is working again!

closed 2020-04-16 by sharaku17

CTCaer, 2020-04-17:
This was fixed in 5.1.4, 0.8.7 already.