CTCaer/hekate #391: Nyx supports Joy Con

chihwa, 2020-04-17:
It would be great Joy Con is supported in Nyx.
It's great for the consoles that have touchscreen issues.

urherenow, 2020-04-17:
Or docked...

MilkiiJugs, 2020-04-20:
It would require drivers just made for that. May add too much consumption. Also, this is made for Issues, not suggestions. Maybe in the future?

CTCaer, 2020-04-20:
Issues here are for feature requests also.

MilkiiJugs, 2020-04-22:
My bad, I will correct that.

fennectech, 2020-04-22:
Something like that is super low on the priority   Ctcaer has much more impactful features     Joycon support is more of a "Would be sweet but may never come"   Theres only so much that can be done in BPMP

CTCaer, 2020-05-01:
https://github.com/CTCaer/hekate/releases/latest

closed 2020-05-01 by CTCaer

chihwa, 2020-05-01:
Wow. It's a surprise that you come up with the implementation this quick! Thank you very much. Big applause to CTCaer and the team! 👏 👏 👏  Thank you very much!

fennectech, 2020-05-01:
Same.   I did NOT see this comming.   There has gotta be some voodoo going on here!!