CTCaer/hekate #392: face a Failed to apply 'nosigchk and Failed to launch HOS issue

zsw12abc, 2020-04-17:
i just download the latest and update my NS to 10.0.0
I try to enter my emuMMC as usual
but face those 2 issues
Failed to apply 'nosigchk
 Failed to launch HOS issue
can you help me?

urherenow, 2020-04-17:
Edit your hekate_ipl.ini and remove the nosigchk=1. There is no patches that work with 10.0 unless you have a modified ams_mitm.kip and loader.kip.

If removing the nosigchk line gets you frozen on a black screen, many things can cause that, to include things that won’t be supported here. In that case, look into the Kosmos package.

closed 2020-04-17 by CTCaer

zsw12abc, 2020-04-17:
[config]
autoboot=0
autoboot_list=0
bootwait=1
verification=1
backlight=100
autohosoff=0
autonogc=1

{AtlasNX/Kosmos w/ nosigchk}
{}
{Discord: discord.teamatlasnx.com}
{Github: git.teamatlasnx.com}
{Patreon: patreon.teamatlasnx.com}
{Pegascape DNS: pegascape.sdsetup.com}
{}

{--- Custom Firmware ---}
[CFW (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
kip1=atmosphere/kips/*
{}

[CFW (EMUMMC)]
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
kip1=atmosphere/kips/*
{}

{--- Stock ---}
[Stock (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
stock=1
icon=bootloader/res/icon_switch.bmp
{}

I didnt see nosigchk=1 in my hekate_ipl.ini

deadair3210, 2020-04-18:
Kip1patch=nosigchk....

Drew-Peacock, 2020-05-24:
I removed "kip1patch=nosigchk" and am still getting the message.

NiniG92, 2020-05-25:
> I removed "kip1patch=nosigchk" and am still getting the message.

open up your hekate_ipl.ini file and just copy + paste the following. should be good to go. :)

[config]
autoboot=0
autoboot_list=0
bootwait=1
verification=1
backlight=100
autohosoff=0
autonogc=1

{AtlasNX/Kosmos w/ nosigchk}
{}
{Discord: discord.teamatlasnx.com}
{Github: git.teamatlasnx.com}
{Patreon: patreon.teamatlasnx.com}
{Pegascape DNS: pegascape.sdsetup.com}
{}

{--- Custom Firmware ---}
[CFW (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
kip1=atmosphere/kips/*
{}

[CFW (EMUMMC)]
fss0=atmosphere/fusee-secondary.bin
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
kip1=atmosphere/kips/*
{}

{--- Stock ---}
[Stock (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
stock=1
icon=bootloader/res/icon_switch.bmp
{}

Retroboy85, 2020-06-15:
Im trying to update my setup as im on 10.2 and i keep getting this error and no matter what i try i cant get noesx to boot, i keep getting this error.

deadair3210, 2020-06-15:
Don't use sig patches

HugoAvila1, 2020-06-26:
> open up your hekate_ipl.ini file and just copy + paste the following. should be good to go. :)
>
> [config]
> autoboot=0
> autoboot_list=0
> bootwait=1
> verification=1
> backlight=100
> autohosoff=0
> autonogc=1 [...]

I just followed this step and now it works, but TINFOIL doesn't work now.... what should I do?

Masamune3210, 2020-06-27:
Tinfoil actively refuses to work if you use Kosmos or Hekate, you will have to figure out how to remedy it yourself as it's not a issue with hekate

CTCaer, 2020-06-27:
No piracy discussion..