CTCaer/hekate #394: Force disable of emummc even in payload chainloading

petrusd10s, 2020-04-20:
Context:

Let's say you have an emunand already in place and activated, and you have to chainload a payload to make it work instead of fss0 method (example being fusee-primary.bin)

[Stock]
payload=payloads/fusee-primary.bin
stock=1
emummc_force_disable=1
icon=bootloader/res/stock.bmp
{}

Would not work because the emummc_force_disable arg is only applied to fss0 method.
And if you use the "payload=x" method you can only use CFW (EMUMMC) and Stock (SYSNAND) being imposible to use CFW (SYSNAND).

This problem can be avoided if you disable the emummc manually from emummc menu, and now the payload would not see any active emummc and would just boot CFW (SYSNAND). But this ain't practical.

My suggestion is to change the emummc_force_disable=1 arg to support even payload chainloading method, or to have some new args to switch or turn off emummc from launch config

Examples being:

emummc_disable=1
emummc_active=RAW1
emummc_active=RAW2
etc...

urherenow, 2020-04-20:
Just stop. When you load a different payload, you are no longer using hekate, and anything in the hekate_ipl.ini is irrelevant. Go bug whomever made the other payload.

CTCaer, 2020-04-20:
Well, just use fss0 method to boot. It is as simple as that.

hekate only keys will always be hekate only keys. Like they are 2 years now.
So, I won't change that behavior, like how I don't mess with any expected behaviors of payloads like fusee-prim or others.

Plus that I won't ever add something that writes files in sd card in every boot.

closed 2020-04-20 by CTCaer