CTCaer/hekate #397: Backup eMMC RAW GGP error

ChantalMo, 2020-04-26:
If I try to make a backup I get this error:

SD & eMMC data (@LBA00FD8000) do not match!#FF0000 verification failed..#FFDD00 please try again...#FFDD00 Failed!

How can I fix this??

Thanks in advance

urherenow, 2020-04-26:
I know you aren’t going to want to hear this, but do you have a different sd card you can try? I just made 2 nand dumps today with zero issues. Using the latest release here.

Backup your card, and run H2testw on it.

ChantalMo, 2020-04-26:
Oké thanks you!
Then I go try another SD card.

ChantalMo, 2020-04-29:
With the new micro sd card is the problem solve.

The other sd card has multiply corrupt sectors. because of this the micro sd card don't work

ChantalMo, 2020-04-29:
Thank you for the tip and help

closed 2020-05-01 by CTCaer