CTCaer/hekate #402: cal0blank=1

deejay87, 2020-05-01:
Setup atmosphere with exosphere.ini
``` [exosphere]
blank_prodinfo_emummc=1
```
Try with setup in hekate_ipl.ini
`cal0blank=1 `

Launch CFW(EMUMMC)
**Result**
Don't change the serial number
(Work if i start fusee-primary)

CTCaer, 2020-05-01:
Redownload the release zip.

This was fixed with https://github.com/CTCaer/hekate/commit/da8c040c082ec4fc23bab6327400075508fb5646
I forgot to push that change and without that, sd card is getting unmounted and then it tries to reach the exosphere.ini and it fails.

You might need to replace your reboot_payload.bin also or where else hekate is copied

deejay87, 2020-05-01:
i have test again with the new hekate_ctcaer_5.2.0.bin and it's the same

CTCaer, 2020-05-01:
What did you try exactly?
show me your hekate ini

deejay87, 2020-05-01:
```
[config]
autoboot=0
autoboot_list=0
bootwait=1
verification=1
backlight=100
autohosoff=0
autonogc=1
cal0blank=1

{--- Custom Firmware ---}
[CFW (EMUMMC)]
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
kip1=atmosphere/kips/*
{}

```
and exosphere.ini
```
[exosphere]
blank_prodinfo_emummc=1
```

deejay87, 2020-05-01:
oh, 
i have make a fresh install with 2 emummc and now it's ok,

CTCaer, 2020-05-02:
have in mind that the keys in exosphere.ini do not interchange.
If you **only** have `blank_prodinfo_emummc` it doesn;t work for sys and if you **only** have `blank_prodinfo_sysmmc` it doesn't work on emummc.

hrvylein, 2020-05-02:
Same here, prodinfo isn't cleared on sysmmc when `[exosphere]
blank_prodinfo_sysmmc=1` is set. I did not set `cal0blank=1` in hekate_ipl.ini as I want atmosphere to take care of it. I've used sdsetup to update my homebrew and double checked that hekate payload is the newest re-release in all folders and when injecting the payload.

This is my hekate_ipl.ini
`[config]
autoboot=0
autoboot_list=0
bootwait=1
verification=1
backlight=100
autohosoff=0
autonogc=1

{AtlasNX/Kosmos w/ nosigchk}
{}
{Discord: discord.teamatlasnx.com}
{Github: git.teamatlasnx.com}
{Patreon: patreon.teamatlasnx.com}
{Pegascape DNS: pegascape.sdsetup.com}
{}

{--- Custom Firmware ---}
[CFW (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
kip1=atmosphere/kips/*
{}

[CFW (EMUMMC)]
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
kip1=atmosphere/kips/*
{}

{--- Stock ---}
[Stock (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
stock=1
icon=bootloader/res/icon_switch.bmp
{}
`

hrvylein, 2020-05-02:
Same here, prodinfo isn't cleared on sysmmc when `[exosphere]
blank_prodinfo_sysmmc=1` is set. I did not set `cal0blank=1` in hekate_ipl.ini as I want atmosphere to take care of it. I've used sdsetup to update my homebrew and double checked that hekate payload is the newest re-release in all folders and when injecting the payload.

This is my hekate_ipl.ini
`[config]
autoboot=0
autoboot_list=0
bootwait=1
verification=1
backlight=100
autohosoff=0
autonogc=1

{--- Custom Firmware ---}
[CFW (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
kip1=atmosphere/kips/*
{}

[CFW (EMUMMC)]
fss0=atmosphere/fusee-secondary.bin
kip1patch=nosigchk
atmosphere=1
logopath=bootloader/bootlogo.bmp
icon=bootloader/res/icon_payload.bmp
kip1=atmosphere/kips/*
{}

{--- Stock ---}
[Stock (SYSNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
stock=1
icon=bootloader/res/icon_switch.bmp
{}
`

deejay87, 2020-05-02:
it's exactly what i have write in the setup i show you, i'm going to try today and continue somes tests :+1: 
I understand exosphere.ini work on launch fusee-primary and setup cal0blank=1 do the job with launch > cfw(emummc), i'ts the tests i do, and control the serial is with 0000000000

closed 2020-05-02 by deejay87