CTCaer/hekate #403: How to install L4T and Android via Hekate?

death7lord, 2020-05-01:
Hello. 
How to install L4T and Android via Hekate? I'm sorry, I didn't understand. Both image has *.img extensions.

CTCaer, 2020-05-01:
You can't install oreo or current images.
The supported images will tell you that they support that installation method

OldGnashburg, 2020-05-01:
What do you mean? Pie hasn't been released yet and L4T Ubuntu is the only other publicly released Switchroot OS.

death7lord, 2020-05-01:
**CTCaer**. Is it method for a future L4T and Android releases? Not for a current images?

death7lord, 2020-05-01:
> The supported images will tell you that they support that installation method
**CTCaer**. Is it method for a **_future_** L4T and Android releases? Not for a current images?

CTCaer, 2020-05-01:
Yeah it's for the L4T images that are coming soon and for if you know how to compile Pie

death7lord, 2020-05-01:
Before I used SwitchSDSetup for it. https://github.com/lulle2007200/SwitchSDSetup

CTCaer, 2020-05-01:
Hekate installation doesn't write the whole sd with a disk image.
It writes a partition image. That's the difference and why current images will not work with that.

People are already hard working to bring support for that. Just be a bit patient.
I had to release today, otherwise I would miss the anniversary again. And I didn't want to disable these 2 features and make a new release in a couple of days.

Existing methods can still work with the disk images. Hekate method will be with partition images.

death7lord, 2020-05-01:
_Don't worry, we are patient =)_

OldGnashburg, 2020-05-01:
Should this be closed then? Not really an issue ATM. SwitchSDSetup in a Linux VM or Live CD is the best method for current public images.

death7lord, 2020-05-01:
Ok. I'll close.

closed 2020-05-01 by death7lord