CTCaer/hekate #404: Partition Manager : emuMMC (RAW)

deejay87, 2020-05-01:
Hi
I'll do somes tests :
Don't know if it's an issue but can't create a partition for emumcc , only linux and android can be set with the slider

CTCaer, 2020-05-01:
If you have a small sd card < 32GB, that's why probably. Or you didn't move the slide to middle place.

deejay87, 2020-05-01:
no it's 128go

CTCaer, 2020-05-01:
Did you put the slider to the middle?

deejay87, 2020-05-01:
the red slider don't move

CTCaer, 2020-05-01:
Well it's not a bug. It works like that.
You need to catch the red and move it to middle.
If you leave it before middle, it will return back.

I will check if I can mitigate that, but don't expect anything. This is how the GUI lib works.

deejay87, 2020-05-01:
yes but like i say you can't move at start, you have to clic to middle directly

CTCaer, 2020-05-01:
Again, that's how it works. **It doesn't move**.
It **only** moves when your finger reaches the middle.

deejay87, 2020-05-01:
ok thanks :), i continue to test

CTCaer, 2020-05-01:
Catch the red circle with your finger and move it left and right without removing you r finger.
You will get what I mean.

deejay87, 2020-05-01:
yes i have understand, have to slide very large to the middle and it move, now i'm testing install of the 2 emummc now

deejay87, 2020-05-01:
Perfect 2 emunand is ok :+1:

CTCaer, 2020-05-01:
>yes i have understand, have to slide very large to the middle and it move

Yeah exactly

deejay87, 2020-05-02:
Just one question to finish this perfect test, possible to setup on hekate_ipl.ini to launch RAW1 and RAW2 here, and not going on "Change emuMMC"
Thanks

CTCaer, 2020-05-02:
@deejay87 That was supposed to work on 5.2.0 with the `emupath` key.
But I have a bug and it doesn't work.

I'm currently collecting bugs, so I can fix them in about 3 days.

deejay87, 2020-05-02:
oki perfect, when it's done i test, have one sdcard ready :+1:

closed 2020-05-05 by CTCaer