CTCaer/hekate #406: Hekate 5.2 / Nyx 0.9.0 MAY not be compatible with 400+gb sd cards?

Clydefrosch22, 2020-05-02:
Hey there.
At first, I thought this was an issue with my sd card and it just having broken, but after trying various things, I believe the issue might actually be with the newest Hekate/nyx.

So I'm still on HOS 9.1 with a fat32 formatted 400gb sandisk sd that worked fine for over a year now.
I updated my Kosmos files including hekate and nyx and when I was done, tried to boot into cfw as usual, but after the sept splash, I got "a fatal error occured when running atmosphere" as an error message, this title id 010041544d530000, a message about saving an error log to sd and to reboot with the Power button.
I do and get more errors. Hard resetting and trying to boot to hekate only brings me to the old, non gui version.
On PC, the sd cards file system isn't recognized anymore, recovery of several kinds not working, so ultimately, i format the card and start over.
same thing happens again.
the card has to be working fine while entering hekate, as it loads the gui from sd card just fine, but when trying to boot to cfw, the error occurs (after several more tries, sometimes there's no error and just a black screen) and the sd card is unrecognizable again.

I've reformatted, checked for errors, loaded up about 100gb of files and back without issue on the sd card.
A 32 gb sd card using the exact same files also loads into atmosphere fine.
And the 400gb card using my old cfw files with hekate 5.1.3 (which i was lucky enough to had backed up before starting this) also gets back into atmosphere without the error.

Now, the issue could very well also be in the newest atmosphere, or in a combination of atmosphere and hekate with this large sized sd card, but i'm mostly postitive it's not just hardware error on my side and if anyone could look into it, that would be grand.

Thank you

CTCaer, 2020-05-02:
Hey,

Can you find me in retronx discord, so I can debug that and see what's happening?

CTCaer, 2020-05-02:
So after debugging the issue, the reason is that the card has no MBR.
FAT32 formatter.exe tool does not find any and it creates an SFD partition scheme.

That's not compatible with numerous stuff and it's an archaic partition scheme.

The fix is to use a tool that can create the MBR and then format it to FAT32.

fennectech, 2020-05-02:
I recomend using gparted on a Linux box or on windows there is a way to do it in disk management

CTCaer, 2020-05-02:
Any partition manager tool will do a good job.
Also new Nyx can do that if you choose to have only the FAT partition.
But not if you can't access it.

closed 2020-05-05 by CTCaer