CTCaer/hekate #408: linux flashing seems to be harcoded

vpeterhun, 2020-05-02:
Hi,

I was trying to flash linux, couldn't figured out, looked at the source.
It seems like the linux image path is hard coded to be "switchroot/install/l4t.00"
I tried it with that path, now it works.

CTCaer, 2020-05-02:
Yes they are and you can find them in `Note 3` inside the partition manager.

closed 2020-05-02 by CTCaer

vpeterhun, 2020-05-02:
Okay thanks, I guess I'm not bright enough to understand the note

CTCaer, 2020-05-02:
The note effectively says that the Flash options use the directory `switchroot/install/` and read specific filenames in there.
For linux it reads l4t.00, l4t.01, etc. Depending on how big is that distro's rootfs.

vpeterhun, 2020-05-02:
Yes, now I understand. I don't want to take more of your time.
Initially I thought it meant that XX is a wildcard of any kind. I tried "l4t.img", tried the whole img file with "l4t." prefixed in front of it. Then I tried "l4t.XX" as a filename.
Since I had formatted my card to exFAT it didn't occured to me that XX means parts of the file for fat32.
After that I looked at the code.

Now I'm just trying to figure out how to make it "see" the flashed linux, as I can't seem to be able to boot into it.

fennectech, 2020-05-04:
for seeing the flashed linux   Take the contents from the fat32 partition on the switchroot (or l4t-arch) and drop them on your sdcard