CTCaer/hekate #410: [feature request] extend option to update reboot_payload.bin to when user uses "payload="

evertonstz, 2020-05-02:
Currently this option is only supported for the "fss0=" option, users using hekate to chainload other payloads could also benefit from this feature.

CTCaer, 2020-05-02:
The reason is that it checks its location. So that can't be used with a payload.
If you need you features of hekate you need to use hekate to boot.

There's no way to validate payloads and decide if the file should be changed or not.

evertonstz, 2020-05-02:
Are we talking about the same thing? I suppose this new feature will actually replace `reboot_payload.bin` with a hekate's .bin so the Reboot to Payload homebrew would be able to actually reboot to the latest hekate's version that was injected, right? 
If that's the case, why not just use any type of checksum to check if `reboot_payload.bin` is actually the latest hekate's bin?

Or maybe I got confused about what this feature actually is?

CTCaer, 2020-05-02:
That feature is not new.
hekate wasn't ever hardcoding a CFW's main path and it will not start now.
The path of `reboot_payload.bin` is taken from fss0 automatically.

As I said before, `payload=` means nothing special to hekate.
It only means that it's time to chainload to a whatever rcm payload, nothing else.

closed 2020-05-02 by CTCaer