CTCaer/hekate #411: Controller dead zones

designgears, 2020-05-03:
Option to adjust the dead zones to mitigate drift would be awesome.

urherenow, 2020-05-03:
But that’s part of the system already? 

https://en-americas-support.nintendo.com/app/answers/detail/a_id/22340/~/how-to-calibrate-the-controllers#DT:t1-q1a1EP:t1-q1a1-c

CTCaer, 2020-05-03:
Hmm, is it so bad?
I always thought that the current dead zone was already big.
This is possible, but I won't put it as an exposed setting yet (it will be in nyx.ini only).

And probably I won't apply it on USB Gamepad (so it can be calibrated more precisely in PC).

designgears, 2020-05-03:
Not inside HOS, handling it from the PC side is fine for gamepad usage. 

When using hekate it initially seemed alright, but it got pretty annoying. Mine personally drifts to the right and up, making it nearly impossible to move it down and left back to the menu, ended up just using the touchscreen.

CTCaer, 2020-05-03:
Have in mind that the calibration I do is based on current values. So if you press it while it's calibrating, it will drift. Something like the old consoles.

Only that in hekate you can remove the left or right joycon (depending on what drifts) and it will recalibrate.

It also recalibrates every time it loses connection (for example after a backup/restore/ums/emummc creation etc happens).

Make sure that this is not the issue. If it's really the sticky slow return drift type, then yeah a bigger dead zone might not fix it.

The dead zone in hekate is that simple thing
```c
jc_drv_ctx.cx_max = jc_pad->lstick_x + 0x72;
jc_drv_ctx.cx_min = jc_pad->lstick_x - 0x72;
jc_drv_ctx.cy_max = jc_pad->lstick_y + 0x72;
jc_drv_ctx.cy_min = jc_pad->lstick_y - 0x72;
```
A 114 (0x72) units buffer.

designgears, 2020-05-03:
Hit the nail right on the head, I'm grabbing the stick before it finishes calibrating, if I wait until after they rumble everything works great.

CTCaer, 2020-05-04:
Ok then I will not put that. Except if someone asks it in the future and it actually helps.
I'm preparing for a hotfix version now, and I will add a small FAQ in the release notes for questions users had. I might remind there how this works on boot/joycon removal.

closed 2020-05-04 by CTCaer