CTCaer/hekate #412: cal0blank=0 not overriding exosphere.ini's blank_prodinfo_sysmmc=1

Goma1337, 2020-05-03:
Title's self-explanatory. 

I'm trying to set up a boot config that won't blank the cal0 when booting the SysMMC through hekate, but I want fusee-primary boots to do. Exosphere.ini is set to blank both the sysmmc and the emummc.

As such, I have:

[AMS 0.12.0 (Fusee)]
payload=bootloader/payloads/fusee-primary.bin
icon=bootloader/res/icon_ams_fusee.bmp

[AMS 0.12.0 (SysNAND)]
emummc_force_disable=1
fss0=atmosphere/fusee-secondary.bin
cal0blank=0
icon=bootloader/res/icon_ams_sysnand.bmp

When I boot into the second option, the cal0 is still blanked, meaning exosphere is not being overridden.

CTCaer, 2020-05-03:
Hmm that's weird.
The code effectively does:

`(ctxt->exo_cfg.cal0_blank && *ctxt->exo_cfg.cal0_blank)`
or
`(!ctxt->exo_cfg.cal0_blank && cal0_blanking)`

Which in your case is:
`true && false`
or
`false && true`
which equals as:
`false`

You sure that you haven't blanked your prodinfo another way?
If you change `blank_prodinfo_sysmmc=1` to `blank_prodinfo_sysmmc=0`, what happens?

CTCaer, 2020-05-03:
Never mind, I re checked the code again and I found the issue.

Goma1337, 2020-05-03:
It's blanked out definitely (incognito_RCM) for the emummc, but not for the sysmmc, as I use the sysmmc to play online.

Oddly enough, when I make blank_prodinfo_sysmmc=0, the prodinfo is still blanked. When I remove cal0blank=0 entirely, it shows up again.

Goma1337, 2020-05-03:
Oh, awesome! Was just typing the answer out when you replied you found the bug :)

CTCaer, 2020-05-03:
Ye thanks for reporting it. The variable was intended to be a u8 at first but I changed it on the refactoring without fixing the allocation.

I'm preparing a new version that will have all fixes for bugs reported for the huge update.
This will happen tomorrow or tuesday. So please use `blank_prodinfo_sysmmc=0` for now.

CTCaer, 2020-05-05:
Fixed in v5.2.1.

closed 2020-05-05 by CTCaer

Bossiskeyboard, 2023-10-27:
this issue is back in the latest version for me

urherenow, 2023-10-27:
I'm not able to load github at the moment, so I can't see this whole issue. But I must ask... are you using package3 or fusee?

gnxswitch, 2023-10-27:
Right now it's working as intended for me, but I have cal0blank activated on Hekate and also on exosphere, booting with package3. Not sure if another combination might be causing issues, but my setup is working fine.

Bossiskeyboard, 2023-10-29:
nevermind