CTCaer/hekate #413: corrupted sd card

meysam25, 2020-05-03:
injecting hekate_ctcaer_5.2.0 payload make my sd card to be corrupted
i used this site for create package 
[](https://sdsetup.com/console?switch#atmosphere;atmos_musthave;atmos_bootlogo;kosmos_toolkit;kosmos_updater;tesla_menu;atmos_ovlloader;atmos_emuiibo;atmos_sys-clk;atmos_sysftpd;atmos-sys-con;ldn_mitm;ovl_sysmodules;status_monitor_overlay;appstorenx;hbmenu;edizon;goldleaf;nxdumptool;lockpick;ppsspp;hekate;lockpick_rcm;)
and i tested on fw 9.2.0

fennectech, 2020-05-03:
Can you test your card with h2testw?

deejay87, 2020-05-03:
Inject payload can't do this, i help a lot and i've never see that.
Remember lot of people have problems with sdcard, it depend if it's a brand mark or not (exemple with samsung evo who have big problems)

Clydefrosch22, 2020-05-03:
In case it happens like it happened to me in this issue, with windows not recognizing the file system and demanding it to be reformatted:
https://github.com/CTCaer/hekate/issues/406

CTCaer helped me fix it by using MiniTool Partition Wizard (careful, it'll ask you if you want to install all kinds of other junk).

the steps in minitool were 
1. Select your SD Card from the list at the bottom (and be very careful to not select your other drives)
2. in the left menu select delete partition, check again that this is your sd card and no other drive, then apply
next is rebuild MBR, apply again
finally, select create new partition (fat32 format , 32k) and apply again.

afterwards, put kosmos/atmosphere/whatever on the card and hekate shouldn't break anymore.

meysam25, 2020-05-03:
all my data has been lost , i am installing game again
i inject this payload (hekate_ctcaer_5.2.0) 8 time and every time my cd card become corrupted i test this in fw 9.2.0 and 9.1.0.
the payload is successful and after payload finished , kosmos and atmospher starting to booting up .
after changing 3 splash screen it will show me error message, i dont remember error message but its say something like : some error happened and they save log into my sd card and they show me the path that log is saved but when i plug sd card to my computer its not recognizing sd card.
my sd card is 256 exfat, i dont remember why i have some problem and i have to make it exfat.

deejay87, 2020-05-03:
don't use Exfat, Fat32 is the best , it's the big error you can do, this create lot of corrupt data
and test your carte too, h2testw can tell you that
if you don't know how to build yourself atmosphere/hekate/patches, take the last sdsetup , not need to discribe

Clydefrosch22, 2020-05-03:
Yes, this sounds exactly like what happened to me.
Old setup worked, as did other sd cards. 

The issue, ultimately, is with the sd card and it lacking this MBR.

Using minitool partition wizard and following those steps above should fix this and allow you to use the newest hekate without corruption.

Back up your nintendo folder so you dont have to reinstall the games you already reinstalled.

@deejay87 in this case, exfat is very very likely not the issue, and for most people that don't use homebrew like retroarch, exfat corruption barely ever happens.
but if all thats stopping you from using fat32 is the filesize limit, look into usb installation.

meysam25, 2020-05-03:
ok i will test this and let you guys know if its work

meysam25, 2020-05-03:
dude you are great, 
this steps solve my problem 

    Select your SD Card from the list at the bottom (and be very careful to not select your other drives)
    in the left menu select delete partition, check again that this is your sd card and no other drive, because this will completely erase that drive and everything in it. if you're sure, click apply
    next is rebuild MBR, apply again
    finally, select create new partition (fat32 format , 32k) and apply again.

tanx bro

meysam25, 2020-05-03:
i was pretty sure that your solution will not work , but i was wrong

closed 2020-05-03 by meysam25

unbranched, 2020-06-29:
@CTCaer This happened to me too, I think it should be more documented at least to avoid heart attacks to people.

I managed to save my microsd without losing any data.
What is wrong is that the boot superblock of the microsd gets corrupted for some reason every time Hekate is launched, this is why you won't be able to read the microsd contents with any device.
Replaced the corrupted superblock with the backup one, and the card returns usable.

The solution posted above is not what I want, as I prefer to still use exfat. So what really are the requirements?

I'm using fusee-launcher until this won't be fixed.

CTCaer, 2020-06-30:
Well you didn't analyze Clydefrosch22's solution.
Otherwise you would have changed the last step to your needs.
(Also FAT32 is better to be used with a 64KB cluster in the case of switch - big files)

The key is step 3. You need a MBR.
Your card is currently formatted like it's a floppy disk. (SFD).

- Backup your files.
- Follow these steps with a PC based partition manager.
or
- Format via hekate's partition manager and then format back to exFAT.

Also you probably already know that exFAT is really not needed on switch because of the archive bit concatenation and that is also prone to corruption, so I won't bother to change your mind.

NonaSuomy, 2021-09-08:
In case this helps someone else. I had my card's boot sector get corrupted after ejecting and putting it back in when I was trying to upgrade to a larger card.

I first dd the whole card and then just the partitions separate for a sanity check to another drive in case it totally killed it...

```
lsblk 

NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS
sda           8:0    0   3.6T  0 disk 
└─sda1        8:1    0   3.6T  0 part 
sdb           8:16   1  58.9G  0 disk 
├─sdb1        8:17   1  29.7G  0 part 
└─sdb2        8:18   1  29.2G  0 part 
```

Note in Linux I was still able to see the FAT32 partition and mount it but not in Windows it just kept telling me to format it. (Do not format it!)

**Backup Card**

sudo dd if=/dev/sdb conv=sync,noerror bs=64K status=progress > Switch_SD_backup_full.bin

sudo dd if=/dev/sdb1 conv=sync,noerror bs=64K status=progress > Switch_SD_backup_FAT32.bin

sudo dd if=/dev/sdb2 conv=sync,noerror bs=64K status=progress > Switch_SD_backup_emuMMC.bin

**Fix Boot Sector**

 `sudo fsck.vfat /dev/sdb1 `                                  

```
fsck.fat 4.2 (2021-01-31)
There are differences between boot sector and its backup.
This is mostly harmless. Differences: (offset:original/backup)
  0:ff/eb, 1:ff/fe, 2:93/90, 65:01/00
1) Copy original to backup
2) Copy backup to original
3) No action
[123?q]? 2
```

```
*** Filesystem was changed ***
The changes have not yet been written, you can still choose to leave the
filesystem unmodified:
1) Write changes
2) Leave filesystem unchanged
[12?q]? 1
/dev/sdb1: 1428 files, 482844/486400 clusters
```

If you don't get this message "There are differences between boot sector and its backup." 
after running fsck.vfat this probably won't help you.

GavinDarkglider, 2021-09-08:
> Inject payload can't do this, i help a lot and i've never see that.
> Remember lot of people have problems with sdcard, it depend if it's a brand mark or not (exemple with samsung evo who have big problems)
> And fat32 is recommended to avoid corrupted files

Just about everything you said here is wrong..... we recommend samsung evo cards for their random R/W speeds, we recommend fat32 to prevent corrupting sdcard with uboot/linux because hos driver breaks partition after linux/uboot touch it. That being said, injecting the payload cant cause filesystem errors. you can inject the payload without the sdcard being inserted.
Only time I have seen issues like this are bad connections on the sdcard connector on the motherboard, which will act like the sdcard was pulled from the console, which can cause corruption.