CTCaer/hekate #415: Suggestion: Put USB-access on the main screen/allow to cancel out of it while its waiting for a connection

Clydefrosch22, 2020-05-03:
Alright, the first one is kind of a nobrainer, being able to mount the sd card without taking it out is amazing  and a godsend and I'm absolutely positive that it'll be the second most used feature of hekate past all the launching stuff from here on out.

And while it's just three short taps away, SD card mounting really should have a dedicated little icon right there on the main screen for instant use.

My only gripe with it so far is not being able to abort it, even if you haven't connected the switch to a PC yet. 

Being able to safely cut the connection from the switches side after connecting would also be nice, but I feel there might be a reason why that's not possible already, so potentially disregard that.

Cheers and keep it up

r1cola, 2020-05-04:
Yeah, he was talking about maybe adding in a wait counter for that pop-up IIRC...currently I believe it just waits indefinitely...

bottmint, 2020-05-04:
It states in the release notes that you can cancel the wait by pressing both volume buttons. 
Safely disconnecting from the host side before disconnecting the cable from the switch is there to hopefully prevent people from unsafely unmounting the sd card in the middle of an I/O operation, which can cause issues on your sd card. It's the same as safely ejecting a flash drive, or other removable media. You should always safely eject from the host.
> (Keep in mind that if anything goes wrong or you get stuck during a USB op, you can press both VOL buttons to bail out.)

CTCaer, 2020-05-04:
As bottmint said, you can bail out from every UMS op with VOL- and VOL+.
Just make sure you know what you are doing.

As for ums on home screen, this is a hard no for now. I dislike crowded home screens.

On the other hand, don't forget that you can actually have a HOS homebrew app that can boot directly to your selected UMS option.
Relevant read: https://github.com/CTCaer/hekate/commit/abcb1dec4644e6fa77ab9b70e7e201e336d2d9f1

Clydefrosch22, 2020-05-04:
Ah, my bad, I didn't read far enough into the release notes to notice the Volume button solution. That'll work well enough for me.
Though you might still want to make the Close button functional before cables are connected somewhere down the line, cause if there's one thing people are less likely to read than current release notes, it's previous release notes.
Graying it out after cables are connected is a good way to make secure ejection more likely to happen though.

And fair enough about the homescreen. But there _is_ some prime empty space for a small sd card icon right above the usb slot. Maybe in the future, when even more functions force you to redesign the menu anyways ;)
Anyways, I guess this can be closed then.

closed 2020-05-04 by Clydefrosch22