CTCaer/hekate #416: Nothing appearing on screen when injecting Hekate.

TheRenegadier, 2020-05-04:
I was given a Nintendo switch which is dead. It goes straight to the blue screen when powered on. I found out that it is an unpatched unit. Just to make sure that the RCM and payload injection works, I followed some steps which brought up a QR code on the screen. I'm guessing it's ok. So I found a very detailed video on youtube in which the guy describes how to remedy the blue screen issue by rebuilding the NAND. I thought I'd give it a shot.

I downloaded all the files and got two new memory cards. 1 Sandisk 64GB and 1 Samsung Evo Plus 64GB (it's all i could get my hands on). I formatted the Sandisk card as normal using the PC (exFAT) and tried to inject a payload to it so I could back up the NAND. However the payload just won't go on. According to TegraRcmGUI 2.6, the payload injection is successful.

Then I thought that the card might need formatting to FAT32 after doing that I tried again but still no luck. I've tried the other card as well and several different payloads but I always end up with the same issue. I did manage to get some text on the screen a few times. I have tried the latest version of Hekate and Atmosphere.

I have a feeling that this is recoverable as it's clearly not patched, enters RCM mode and I have tested it for the exploit.

Thanks in advance

CTCaer, 2020-05-04:
Does hekate work if you inject without sd card?

TheRenegadier, 2020-05-04:
All it says is:

"[ERROR] Fatal error: Failed to initialise the SD card.

Press POWER to reboot"

masagrator, 2020-05-04:
You are sure you are injecting Hekate? Because Hekate doesn't need sdcard to work

CTCaer, 2020-05-04:
That's not a hekate message

TheRenegadier, 2020-05-04:
I injected the Hekate payload from the Hekate folder "hekate_ctcaer_5.2.0.bin"
Was that not the right one?

CTCaer, 2020-05-04:
`[ERROR] Fatal error: Failed to initialise the SD card.` that's not hekate and also hekate never tells you to reboot. It goes into its menu after some warnings

TheRenegadier, 2020-05-04:
No RCM device detected
RCM Device detected
Invoking TegraRcmSmash.exe with args : "K:\Switch\hekate_ctcaer_5.2.0_Nyx_0.9.0\hekate_ctcaer_5.2.0.bin"
TegraRcmSmash (32bit) 1.2.1-3 by rajkosto
Opened USB device path \\?\usb#vid_0955&pid_7321#5&1cc0e58a&0&4#{aa0dbd45-3117-f331-5c49-76bf65225042}
RCM Device with id 4003FC150000000805633C6401101062 initialized successfully!
Uploading payload (mezzo size: 92, user size: 122339, total size: 188555, total padded size: 192512)...
Smashing the stack!
Smashed the stack with a 0x7000 byte SETUP request!
Payload successfully injected

CTCaer, 2020-05-04:
Did you inject without sd card?

TheRenegadier, 2020-05-04:
Yes, I've done that now. I've re downloaded the Hekate archive 5.2.0, removed and reinstalled TegraRcmGUI. I've injected the payload there is still a blank screen.

CTCaer, 2020-05-04:
Remove and reinstalled tegrarcmgui?

For the last time, did you **remove the SD CARD from your switch**?
I want an explicit answer to that.

If you again tell me something that I did't ask you to do, I will automatically assume that this post is a troll and I will lock the issue.

TheRenegadier, 2020-05-04:
> Remove and reinstalled tegrarcmgui?
> 
> For the last time, did you **remove the SD CARD from your switch**?
> I want an explicit answer to that.
> 
> If you again tell me something that I did't ask you to do, I will automatically assume that this post is a troll and I will lock the issue.

Believe me, i'm not trolling. **The card is nowhere near the switch**. I've reinstalled TegraRcmGUI and I have some yellow writing on the screen now...

Missing LP0 (sleep mode) library!
Missing or old minerva library!
Update your boot loader folder!

An exception occurred (LR 0002400): Undefined instructions

Press any key...

CTCaer, 2020-05-04:
Finally!
That's exactly why I asked you to remove the card. For that message.

Your bm92t chip is toasted. You need to replace it.
Restoring an emmc backup will not help bringing it back.

closed 2020-05-04 by CTCaer

TheRenegadier, 2020-05-04:
Thanks, can these chips be bought online?

CTCaer, 2020-05-04:
Yep you can find them online. Also every repair shop, should also fix them for you if you prefer that.

TheRenegadier, 2020-05-04:
Do you mean this chip?
https://www.ebay.co.uk/itm/New-Replacement-M92T36-Power-Management-IC-Chip-for-Nintendo-Switch-Console/184152074921?hash=item2ae051aaa9:g:8FAAAOSwXSZeMbz~

CTCaer, 2020-05-04:
yes

TheRenegadier, 2020-05-04:
would it be possible to restore the eMMC once the chip has been replaced?

fennectech, 2020-05-04:
@TheRenegadier  You probally wont need to.

TheRenegadier, 2020-05-04:
really? even though its blue screening at the moment?

TheRenegadier, 2020-05-22:
So guys, a bit of an update, the chip has been replaced with a brand new one. The problem still exists. any more ideas?