CTCaer/hekate #418: Black screen when booting Hekate CTCaer_5.2.0.bin

Jayinem, 2020-05-05:
I just updated today from 9.2 to 10.0.2. I can boot both Atmosphere and SX OS. But when I choose Hekate CTCaer_5.2.0.bin I get a black screen. I've googled and haven't really found any solutions to this problem.

Jayinem, 2020-05-05:
Nevermind I'm dumb I didn't think any files on sd card were needed for Hekate. Adding them now.

Jayinem, 2020-05-05:
When I run nyx_usb_max_rate (run once per windows pc) it seemed to make my PC no longer read nxmtp for some reason. That's how I usually access files from my sd card to my PC instead of removing sd card all the time.

CTCaer, 2020-05-05:
The reg file can't affect non Nyx usb devices.
hekate's USB Vendor ID and Product ID are specific.
You probably need to reboot because windows decided to break usb like always.

Correlation does not imply causation. You can edit that file and see that its contents are valid.

Jayinem, 2020-05-05:
I rebooted multiple times and my computer won't read my switch exactly after I installed nyx usb devices. Even tried other apps and it no longer works. Works fine on other computer with the same switch/setup. 

Could you just tell me how to uninstall it please?

r1cola, 2020-05-05:
Nope, all it does is 

> 
> 
> I rebooted multiple times and my computer won't read my switch exactly after I installed nyx usb. Even tried other apps and it no longer works. Works fine on other computer with the same switch/setup. I was using nxmtp right before I did that to transfer atmosphere and firmware 10.0.2 files to my switch with no problems. So I'm sorry but there's coincidences and there's situations where it's pretty obvious what the cause was. I mean it's a usb related .reg and I just happen to not be able to use usb right after I click it?
> 
> Could you tell me how to uninstall it please?
> 
> Also yes I downloaded it the latest hekate from here. If I uninstall it and my switch still doesn't work you're right. I removed the registry key that's in the .reg is there anything else I need to remove?

nope, all it does is add that key to your registry, so once that's gone its done...

Jayinem, 2020-05-05:
Well your guys app messes stuff up and you don't responsibility or try to help. You do a lot of great work but this is not right.

CTCaer, 2020-05-05:
So with your sayings 
>was using nxmtp right before I did that to transfer

That means that nxmtp broke it. But as I said `correlation does not imply causation`.
There's no magic in that reg. Because it can't break anything. If it did then every usb storage you ever bought with would kill your pc's usb.

Your removed and fixed nothing. So here's your proof.

So discussion is over. Especially when you actually don't want help with such behavior here.

closed 2020-05-05 by CTCaer