CTCaer/hekate #419: Partition Manager - error message not fit

Borrajo, 2020-05-05:
In partition Manager when trying to format a 32gb sd with a 27gb linux partition (the maximum allowed) the following error appears. The issue is that the message does not enter the dialog

[Photo 'Partition Manager' Error](https://drive.google.com/open?id=1ImorHfnXz6qd5hi9RWXhvyNMCkhGLUBi)

CTCaer, 2020-05-05:
I'll fix that.
It basically says
 `Remove the SD card and check that is OK.`
`If not, format it and press any key!`

Which also needs a bit of change. Because it expects you to re insert and then press Power to restore files.

I also wonder why that message happened.
(I tried small fat in there and worked, but not with a 32GB card.)

CTCaer, 2020-05-05:
Fixed in v0.9.0.

closed 2020-05-05 by CTCaer