CTCaer/hekate #420: hekate 5.2.0 fix archive bit issue

earandel, 2020-05-05:
I ran into a strange issue with the new archive bit fixing functionality in hekate 5.2.0.  I was trying to update the firmware.  The process that has always worked is to create a folder on computer, extract firmware files to the folder, delete file metadata using xattr (macOS oddity that ends up with ._* files on sd card if not done), move or copy the directory to the sd card, run hekate archive bit fix outside of Nintendo folder, and run choi to update the firmware.  When I tried fixing the archive bits using hekate 5.2.0, it said it completed successfully but when I booted into HOS, choi saw the folder with the firmware files as a file and wouldn't let me navigate to it.  The same was true for NX-Shell.  I tried renaming the folder and rebooted HOS a few times with no change.

I reverted back to hekate 5.1.4 (both payload and boatloader files on sd card) and ran the archive bit fix in both Nintendo and non-Nintendo folders since I was worried that hekate 5.2.0 may have messed something up now that it fixes archive bits across all files/folders on the sd card.  After doing this and rebooting into HOS, the choi update process worked as expected.

CTCaer, 2020-05-05:
What are the names of these folders?

(Also couldn't you report it a bit earlier? There's a new release now. ^^)

earandel, 2020-05-05:
The folder was called NX-10.0.2 when archive bit fixing was run.

I just saw that a new release went out after I posted this issue! :)  I should have jumped on this earlier.

CTCaer, 2020-05-05:
Yeah because it's easy to fix.

Issue is the the NX-10`.0.2`. This counts as an extension. Dot and 3 chars.

For now the way to fix that is to rename it to `NX-10_0_2`

earandel, 2020-05-05:
Makes sense.  Thanks for the quick response!

CTCaer, 2020-05-05:
Hmm. Release has low count of downloads for now. I may update the release.

CTCaer, 2020-05-05:
@earandel fixed with https://github.com/CTCaer/hekate/commit/9698cbc9d71b1f26e79bcb70edc6d136ea043aee

Release zip updated. Please test.

CTCaer, 2020-05-05:
Yep, confirmed fixed.

closed 2020-05-05 by CTCaer

earandel, 2020-05-05:
Reproduced issue on 5.2.0 and tested latest 5.2.1 version to confirm fix worked successfully.  Thank you!

CTCaer, 2020-05-05:
Thanks for confirming.