CTCaer/hekate #421: emupath issue

snoofly, 2020-05-07:
I have 2 emummc partitions created via hekate, RAW1 dirty, RAW2 clean.

In order to boot to a particular one I have to use Change emummc in Hekate.

According to the README my understanding is that I should be able to add bootloader entries in hekate_ipl.ini with emupath=/emuMMC/RAW1 or emupath=/emuMMC/RAW2 to be able to select the emuMMC from the Launch menu and avoid having to change the setting but the emummc that is saved to emummc.ini is always the one that boots.

snoofly, 2020-05-07:
Maybe this is already known and fixed with this commit:
https://github.com/CTCaer/hekate/commit/75432344018c7989ffa373dc647b05553b7c9f57
Please close if so.

CTCaer, 2020-05-07:
This key does not touch the emummc.ini. 
The design of emummc.ini is that it's only to be changed/written via user input.
The emupath= overrides the emummc.ini in ram, basically.

snoofly, 2020-05-07:
Thanks for responding.
So is it possible to setup separate bootloader entries using emupath to 2 emummcs?
That would be a really useful feature for anyone who has set up a clean and dirty emummc and wants to be able to select them via launch options, which is what I thought emupath was for.

CTCaer, 2020-05-07:
Yeah. emupath= is exactly for that case: Using multiple emuMMCs.
example:
```ini
[Emu 1]
fss0=atmosphere/fusee-secondary.bin
emupath=emuMMC/RAW1
emummcforce=1

[Emu 2]
fss0=atmosphere/fusee-secondary.bin
emupath=emuMMC/RAW2
emummcforce=1
```
As you can see, I also added `emummcforce=1`. 
In case something goes wrong or the folders get deleted, this disallows hekate to continue.
Useful when you don't want to automatically boot in sysnand CFW if emummc is not found.

closed 2020-05-07 by CTCaer

Bladen123, 2023-07-27:
Still doesnt work LOL , i created 2 emummcs , different users and apps and still allways loads EMUMMC RAW 1 even in hekate_ipl is set emupath=emuMMC/RAW2. This is a joke , like we could easy get our consoles banned if for example emummc 1 even doesnt work for some reasons like this EMUPATH = doesnt work too