CTCaer/hekate #422: emuMMC backup files name

AD2076, 2020-05-07:
Can you save emuMMC backups as rawemummc.bin.XX instead of rawnand.bin.XX
Users can have tidier backups and Hekate could avoid "improper" restores or at least warn the user.
Restoring sys over emu is not a big issue and can be useful, restoring emu over sys could be nasty.

CTCaer, 2020-05-07:
User should always copy the backup to pc as the whole folder that has it and rename it so they know what it is.

On the other hand, this will eventually be more proper.

hrvylein, 2020-11-25:
What do you mean by "nasty"? Possibility of getting banned or is there any technical reason a emuMMC backup should not be restored to sysnand?

CTCaer, 2022-07-07:
backing up emummc, now uses `/backup/{emmc_sn}/emummc` folder

(Restore still expects it in `/backup/{emmc_sn}/restore`)

closed 2022-07-07 by CTCaer