CTCaer/hekate #423: understanding hekate folder layout.

LordRafa, 2020-05-07:
Hello,

I am trying to use this code to write a payload to read/write my Nvidia Shield emmc. I already ported the fusee gelee to the t124 so I would like to continue doing some work to port linux mainline to it.

After some grep, I have realized that a good amount from the bootloader code is duplicated on the nyx code. I would like to confirm if these are two independent payloads. Also it would be really helpful if anyone could help explaining this software flow. The problem is that this software does much more than I need but I don't even have a Switch to check all its functionality.

My idea would be start the payload, read 1000 bytes from EMMC to IRAM and do a EP1_IN_WRITE then repeat the process till the whole EMMC is dump.... Not sure how slow would this be but it really doesn't matter as long as I can backup my emmc and restore it if I mess something playing with nvtboot...

Kind regards,
Rafael Waldo

CTCaer, 2020-05-08:
Hi.
Yeah the payloads are independent and lots of hw init code is duplicated.
In the next days I plan to refactor all of them and put them in 1 folder (the ones that are the same).

You always need to take care with porting any Switch based payload, because the regulators are configured for this platform and they can burn yours.

If you need the fastest possible way to dump your eMMC then you need to use the usb code (it's currently only in Nyx).
It can allow you to mount your eMMC as a usb mass storage device. Read speeds of that reach 40-41MB/s (almost at the limits of tegra x1 USB2 controller).

LordRafa, 2020-05-08:
Hum, this is something that I was worried about...

So the power/clock initialization of the eMMC is not done already at the point that RCM jumps to the payload, hence I would need to deal with it myself, right?

On the other hand, how is the eMMC detected in the computer? Like a device that doesn't have any format but that can be dump with eg. dd? I am really sorry for this silly question but I am not familiar at al with Switch scene...

CTCaer, 2020-05-09:
Power/Clock init/deini is done on brom when booting normally. On RCM you need to do it on your own. Powering is automatic because it relies on the always on 1.8V rail.
But you still need to comment out any sd or display init that might rely on different regulators.

Switch has nothing to do with that. eMMC hw/protocol does not change.
UMS exposes the whole physical disk. So you can choose to use a partition backup tool or just dd the whole disk.

bylaws, 2020-05-09:
It might be an idea to join the switchroot discord @LordRafa, there is shield tablet dev there who has ran mainline: https://discord.com/invite/G8yCKt3

LordRafa, 2020-05-09:
Hum, thanks for your comments, I will start to remove off anything non related to emmc from your code so I can have something to start from and then I would double check which is the power rail use on the shield to power the eMMC... hopefully it would be the 1.8v as you say and no the 3.3v... otherwise I think that it is going to be really painful without specs...

This also gives me an idea about why I couldn't reenable the [JTAG](https://github.com/LordRafa/ShofEL2-for-T124/blob/t124/payloads/jtag_example.c) on my TK1 just enabling the control regs... I think that probably I would need to start the clock on this iface before can be used but I have no clue how to do it...

I am fully aware that dealing with power/clock domains is something really common on baremetal level but I would need to do some learn about because I never have to deal with it.

Thanks for the invite @bylaws, it will take me some time to process @CTCaer info but I have no doubt that this channel will be really useful in the future...

urherenow, 2020-05-10:
Sounds like you’re well on your way to making a brick. Or maybe I misunderstood what you’re saying.

It doesn’t matter how a shield runs at the bare-metal level, because the Switch is NOT a Shield.  Different hardware. Do not try to change voltages based off of something else.

There is already an Android for Switch, and an .apk that makes it identify as a Shield. In fact (I had an Android back in the Galaxy S2 days) as soon as I set up my Switch, Google sent me a Welcome Back email, letting me know that I could download everything I purchased before, onto my new Shield.

CTCaer, 2020-05-10:
@urherenow Please do not hijack the issue thread without knowing what you are talking about.

The technical discussion here is specific and there are also working versions of hekate for jetson x1 and x1 based shield.

@LordRafa As for the TK1 port, feel free to join that discord that @bylaws said and ping us.
It will not be easy, but it's doable.

LordRafa, 2020-05-10:
Thx @CTCaer it will take me some time to work on this because now I am with the finals but in a couple of weeks I will come back to this. I started to remove everything that is not need for the readrawemmc function and from there I will start to move on.

closed 2020-06-01 by CTCaer